debian-pi/raspbian-ua-netinst #297: copy/paste hassles

pictsidhe, 2015-10-16:
OK, this is a silly one, but I won't be the only one.
Being a lazy and bad typist, I copy/paste a lot. When setting up raspbian-ua-netinst, I copy/pasted installer-config.txt filename when saving my tweaks. When I ran ua-netinst, installer-config was ignored. I stared and stared, but couldn't work out why, maybe everythng needed to be root, so I reflashed ua-netinst and tried to cp installer-config.txt. Error! I eventually determined that the description page '-' was not the '-' that I type, or ua-netinst was looking for, despite looking identical.
Fix: edit description page with the right '-', if that doesn't work, tell us lazy people to type it!

diederikdehaas, 2015-10-16:
Maybe it's a good idea to format various things like `installer-config.txt` but I'm not sure if that will fix it.
But I'm probably not willing to go further then that as I don't want to encourage copy/paste because various issues have cropped up because ppl copy/pasted the defaults from that page **which you should NOT do !**

pictsidhe, 2015-10-17:
It was just the filename I copy/pasted. Not good practice, I admit, but it almost always works fine...

closed 2015-10-17 by pictsidhe

diederikdehaas, 2015-10-17:
Have you tried whether it works to copy/paste with the formatting I used in my previous post?