debian-pi/raspbian-ua-netinst #299: SD partioning should be changed

pictsidhe, 2015-10-17:
The SD card is formatted from the first available block. This is a really bad idea on SD cards. The first erase block (usually 4M) has firmware stuff on it. If it dies, so does the whole card. It isn't wear leveled so is extra fragile. This is why almost all Raspbian images start their first partition at block 8192. I killed a lot of compact flash cards on x86 systems a few years ago before I knew this...

I tried adding variables to rcS to fix this, but while a test busybox script worked on my laptop to fdisk the sd, it failed on a ua build, it sat there and did nothing, no log, either.
I tried this:
bootoffset=8192 used for first boot partition block
rootoffset=$((`echo $bootsize | sed s/+// | sed s/M//` * 2048 + $offset )) used for first rootpartition block when bootdev=rootdev, otherwise it starts at bootoffset

I gave up and just set rootoffset=270336 after a few goes. In hindsight, could make partition 3 which ends at 8191 first, then do /boot and /root, then delete 3. Should work?

diederikdehaas, 2015-10-17:
I have submitted [a PR which replaces busybox-static with busybox](https://github.com/debian-pi/raspbian-ua-netinst/pull/296) and a consequence of that change is (also) that the normal fdisk is used, which _should_ do proper alignment:
```
root@rasppi-2b:/home/diederik# parted /dev/mmcblk0 align-check opt 1
1 aligned
root@rasppi-2b:/home/diederik# parted /dev/mmcblk0 align-check opt 2
2 aligned
root@rasppi-2b:/home/diederik# fdisk -l

Disk /dev/mmcblk0: 14,7 GiB, 15720251392 bytes, 30703616 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xc20fd0ff

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        2048   264191   262144  128M  b W95 FAT32
/dev/mmcblk0p2      264192 30703615 30439424 14,5G 83 Linux
```
Are you saying that `fdisk` is still 'wrong'? 
Part of the partitioning process is also [erasing the MBR](https://github.com/debian-pi/raspbian-ua-netinst/blob/master/scripts/etc/init.d/rcS#L416). Wouldn't that also erase the firmware stuff (assuming you're right, I don't know)?

You may also take a look at #183 #152 #21 and #30 (I think I've got them all now :-P )

diederikdehaas, 2015-10-17:
> In hindsight, could make partition 3 which ends at 8191 first, then do /boot and /root, then delete 3. Should work?

Wouldn't that 'destroy' the part with the firmware stuff on it already?

pictsidhe, 2015-10-19:
It did break it, I spoke too soon, it was merrily doing it's 10 minute install.
It appeared to complete and wrote it's log to /var, but the partition table was incorrect. The installer managed to find an fdisk that still runs on cylinders! Still trying to get round that; -u=sectors froze it, no log.  Once it's installed, fdisk runs on sectors, so I'm assuming that's a newer one. Makes it tedious to work out what is going wrong. I also changed buildroot.sh to start the installer partition at sector 8192, mainly to see if moving the start of the boot partition was going to stop it booting. It didn't, that works fine, if I haven't broken the rcS fdisking bit. I also tried commenting out the fdisk bit and partitioning the card before running the installer. Didn't work, no log. Interrupting one run of that showed it installing to root. I've attempted to get it to copy the log to /boot just before the reboot, but that hasn't helped with no log failures. 
That 1st erase block really shouldn't be used if you want decent life from the SD card. Standard ua-netinst partitioned my 32GB card at sector 16, with a +256M boot partition ending at 500095, which is a metric 256MB? If I fdisk on my laptop, it starts at sector 2048 by default. The 32GB card, when virgin, had it's partition start at sector 8192, which is almost standard these days. I've come across a few that started at 8128 (Kingston). I always check where new cards are partitioned from before doing anything with them now. Small cards might have 1MB erase blocks, but most now have 4MB blocks. Partitions should be aligned to 4MB, unless it's an 8128 card.

I need to get an install done soon, so probably a bloated one now, but will maybe come back to this in a month or so.

pictsidhe, 2015-10-22:
I gave up on this. I did get it to partition as I wanted, but -u=sectors didn't work, u as an fdisk command did. Then it was funny about +128M etc partition size, using smaller sizes with strange boundaries. The only way I got it to put boundaries where I wanted was to use sectors. That put the partitions in the right place. Creating partition 3 as the first fdisk action then deleting it before writing is invisible to the installer as it is isn't written. As partitioned, the system would install itself, then corrupt mmcblk0p2 on reboot. Every time. I could make it readable on my laptop again with fsck, but it still wouldn't boot. I successfully changed the installer to start at sector 8192, which was encouraging. Maybe the standard fdisk instead of busybox-static fdisk will do better? Though commenting out the fdisk bit and manually partitioning the card before installation didn't help, I suspect something else is awry too.

diederikdehaas, 2015-12-09:
@pictsidhe have you tried your experiments against the v1.0.x branch or the v1.1.x ?
The latter (now) uses the 'real' `fdisk` and should thus support the `-u` parameter.

And isn't the only thing you need the `bootoffset` parameter as the default (for root) would be to do it directly after the end of the boot partition? Or am I missing something?

diederikdehaas, 2015-12-25:
@pictsidhe I have just merged #332 into v1.1.x and you should now have the option to specify your own bootoffset. You will have to build the installer yourself off the v1.1.x branch. Let me know if that's a problem.
The default hasn't changed yet, so it is still 2048, but can you verify whether it could fix the issue?

dragetd, 2015-12-25:
I have trouble finding ANY information online about any issues with sector one.

As far as I know, partition tools might not use the first sector because some storage devices report 512-byte sectors while using 4k hardware sectors… causing performance issues if your sector access is not aligned.

Eventually you need to clear up the expression 'firmware' here.
'raspberry pi firmware' - usually the term used for the kernel and boot stuff needed for the raspberrypi to start or
'sdcard firmware' - the firmware controlling the sdcard and wearleveling etc.

The sdcard firmware is usually not accessible through the regular storage a dresses and only through special vendor-specific commands. Storage sector 0 will have the same treatment as storage sector 4096 or whatnot.

The raspberrypi firmware usually does not care where it is… just needs to be loaded by the broadcom chip.

If you are actually reffering to the sdcard firmware... do you have any links or information about this?

diederikdehaas, 2015-12-26:
Link given by @dragetd :+1:  : 
https://www.thomas-krenn.com/de/wiki/Partition_Alignment (in Deutsch) [english translation](https://translate.google.com/translate?sl=auto&tl=en&js=y&prev=_t&hl=en&ie=UTF-8&u=https%3A%2F%2Fwww.thomas-krenn.com%2Fde%2Fwiki%2FPartition_Alignment&edit-text=&act=url)

dragetd, 2016-01-06:
I did some experiments. So far everything concludes that an SD-Card really do not give a damn which logical sector you write to. There is nothing special about the logical sector 0 (which could be any hardware sector due to wear-leveling).
The internal firmware of the SD-Card controller is only accessible through vendor-specific sdio-commands.

The only impact a partition starting at some earlier sector could have is performance loss due to misaligned filesystem-sectors. The fdisk applet from busybox bundled with Debian is suboptimal, but @diederikdehaas will replace it with the actual fdisk next version.

IMHO this issue can be closed.

Mausy5043, 2016-02-07:
@dragetd : :+1: Thanks for the info.

diederikdehaas, 2016-03-02:
With the merge of PR #371 I consider this issue fixed. 
If you disagree, feel free to reopen it and explain why.

PR #371 (2016-03-02): Changed default bootoffset to 8192.
This fixes #299 and possibly fixes #283.
commit 1dd80861 (2016-03-02): Changed default bootoffset to 8192.

This fixes #299 and possibly fixes #283.
closed 2016-03-02 by diederikdehaas