debian-pi/raspbian-ua-netinst #300: Problem with installer‑config.txt

raivisj, 2015-10-21:
I created installer‑config.txt with 2 lines in it:

release=jessie
bootsize=+300M

Not my RP1B nor my RP2 won't find a file ?! Where can bet problem - file is in ANSI encoding

File located in SDcards root

diederikdehaas, 2015-10-21:
Did you type the filename yourself or did you copy/paste it?
Did you have a blank line at the end?

raivisj, 2015-10-22:
thanks for fast replay, I don't have blank line at end, and I typed filename myself ! With raspbian im working in Windows 7 system, I will try to do same thing in Linux, maybe it changes something

diederikdehaas, 2015-10-22:
As far as I know you need to have a blank line at the end. 
It would be interesting to know whether the ANSI encoding makes a difference as I always use UTF-8

raivisj, 2015-10-22:
Ok, I will test it and give some info back

Gurr3, 2015-10-22:
ANSI does not make a difference.
One of the things done in the installer is sanitizing input files with dos2unix.
[Link to code.](https://github.com/debian-pi/raspbian-ua-netinst/blob/master/scripts/etc/init.d/rcS#L58)

diederikdehaas, 2015-10-22:
@Gurr3 have you verified that or do you think that?

Currently `sanitize_inputfile` only converts linebreaks from dos to unix. 
This issue may point out we need to do more.

goranche, 2015-10-22:
if you're not using "special" characters, utf-8 is ascii, encoding doesn't really come into play (especially not ANSI)
a newline at the end of the file, however, is required... as it's required (or at least recommended) for pretty much all text files under *nix

diederikdehaas, 2015-10-22:
> encoding doesn't really come into play (especially not ANSI)

I'm not so sure about that. 
http://stackoverflow.com/questions/701882/what-is-ansi-format
http://www.differencebetween.net/technology/web-applications/difference-between-ansi-and-ascii/
http://stackoverflow.com/questions/700187/unicode-utf-ascii-ansi-format-differences

As [the copy/paste issue](https://github.com/debian-pi/raspbian-ua-netinst/issues/297) seems not to be the cause, we'd have to explore other possibilities.

goranche, 2015-10-22:
yes, the first answer in the first question pretty much confirms what I'm saying  :wink: 

but sure, I might be wrong :innocent:

diederikdehaas, 2015-10-22:
I'll schedule an appointment with an optometrist then.
The first comment on that first answer:
`ANSI does not necessarily have to map to CP1252. It does, however, always refer to the legacy codepage set for the system. This may be CP1252 on western European or US systems but don't count on that.`
Upon which the first answer was edited.

diederikdehaas, 2015-10-22:
But I'm all ears for other ideas as to why such a simple `installer-config.txt` file isn't picked up ...

goranche, 2015-10-22:
yeah, you might...
```This is essentially an extension of the ASCII character set in that it includes all the ASCII characters with an additional 128 character codes. This difference is due to the fact that "ANSI" encoding is 8-bit rather than 7-bit as ASCII is (ASCII is almost always encoded nowadays as 8-bit bytes with the MSB set to 0). See the article for an explanation of why this encoding is usually referred to as ANSI.```

the main point here is ASCII, ANSI encoding is 8 bit, if you're using normal ascii characters (as stated in my first comment, no "special" characters), then ANSI **is** ASCII (MSB is 0), the same goes for UTF-8

diederikdehaas, 2015-10-22:
I didn't comment on the part that `ASCII` is the same as `UTF-8` when no special characters are used, because I agree.
`ANSI` is very similar to `ASCII` and thus _usually_ should not be an issue. But I think it's worth **verifying** before ruling that out.

Gurr3, 2015-10-22:
Is it the filename or the contents of the files that are the problem?

raivisj, 2015-10-22:
No success, content of installer-config.txt:
<pre>
preset=server
packages=nano
release=jessie
bootsize=+300M
</pre>

after bootsize is blank line/newline, that didn't help. Raspbian installer boots up with default values and dont even see installer-config.txt file

diederikdehaas, 2015-10-22:
Can you offer your `installer-config.txt` as a download? Maybe I can find something if I had the actual file (as opposed to just the contents which you already posted)

raivisj, 2015-10-22:
thanks for help, here is the file - http://s000.tinyupload.com/?file_id=08088757352768712611

goranche, 2015-10-22:
you probably have a problem with the filename, instead of the hyphen (minus sign) you probably used a dash or something similar

raivisj, 2015-10-22:
oki, will check again !

raivisj, 2015-10-22:
guys, sorry for this all post :x it seems the problem was in filename as goranche noticed ! After renaming file installer find the file and see the custom configuration :)

diederikdehaas, 2015-10-22:
No problem, glad that it's sorted now :-)

closed 2015-10-22 by diederikdehaas