debian-pi/raspbian-ua-netinst #302: eth0 not found

aristosv, 2015-10-28:
The netinst works just fine on the Pi. But when I unplug the microSD, and plug it in any other Pi, when booting up, I get a message saying that eth0 was not found. So there is no network. I've tried it with 3 Pi's so far and Raspbian only detects eth0 on the Pi it was setup on.

Error message:

Cannot find device "eth0"
Bind socket to interface: No such device
Failed to bring up eth0.

Help?

goranche, 2015-10-28:
might this be a duplicate of #103 ?

aristosv, 2015-10-28:
Thanks for your reply. I've read that post.
To be clear, if I add these two lines in post-install.txt will the problem be resolved?

sed -i -e 's/KERNEL\!=\"eth\*|/KERNEL\!=\"/' /lib/udev/rules.d/75-persistent-net-generator.rules
dpkg-divert --add --local /lib/udev/rules.d/75-persistent-net-generator.rules

Thanks

goranche, 2015-10-28:
just to be sure it's the same issue, remove the file ```rm -f /etc/udev/rules.d/70-persistent-net.rules``` before moving the SD card to a different raspberry (it should be recreated there, so it will not work on the first one anymore)...

if that works, you might try adding ```net.ifnames=0``` to cmdline.txt (and deleting the persistent rules file)...

and you can use [this](https://github.com/debian-pi/raspbian-ua-netinst/issues/103#issuecomment-54517762) for future installs... but do keep in mind that it's a workaround, it changes files (well, one file) that come with system packages

aristosv, 2015-10-28:
Your suggestion worked. 
How can I make sure that the system file that changed, never goes back to its original state?
I'm worried because I'll be using this image on 45 raspberry pi's, and I don't want them to suddenly lose their network connectivity, after an X amount of time.

goranche, 2015-10-28:
adding ```net.ifnames=0``` to cmdline.txt should do the trick (if I understand correctly, that is)

diederikdehaas, 2015-10-28:
> How can I make sure that the system file that changed, never goes back to its original state?

By using `dpkg-divert` as described [here](https://github.com/debian-pi/raspbian-ua-netinst/issues/103#issuecomment-58824684)

goranche, 2015-10-28:
> By using dpkg-divert

I wouldn't really do that if at all possible, as it requires changing a "system" library script, and then marking it as "non-upgrade-able" (yeah, not a word, I know)... there is potential for much pain later in time

goranche, 2015-11-01:
@aristosv are you ok now, can we close this issue? (considering #103 is still open, and will probably remain so)

aristosv, 2015-11-01:
yes, thank you

closed 2015-11-01 by aristosv