debian-pi/raspbian-ua-netinst #303: Static IP

borpin, 2015-10-29:
Not really an issue but how do I need to change the installer-config.txt so that it installs to a static IP address please?

goranche, 2015-10-29:
just setting the values described in [Installer Customizations](https://github.com/debian-pi/raspbian-ua-netinst#installer-customization) to your liking should do the trick
  * ip_addr=192.168.0.2
  * ip_netmask=255.255.255.0
  * ip_broadcast=192.168.0.255
  * ip_gateway=192.168.0.1
  * ip_nameservers=192.168.0.1

don't forget to use the right filename, use the "minus sign" in the filename... and place a newline at the end of the file (or a complete empty line)

borpin, 2015-10-29:
Thanks.  I think this is one of the Pis best kept secrets.

closed 2015-10-29 by borpin

goranche, 2015-10-29:
> Thanks

you're welcome :wink: