debian-pi/raspbian-ua-netinst #304: thank you - this is awesom

andrewufrank, 2015-11-03:
exactly what i needed for a long time. 
is this portable to, say, install another ARM board (say cubieboard, cubietruck or odroid)?
thank you a lot!
andrew

goranche, 2015-11-03:
glad you like it :+1: 

regarding the other boards: YMMV, but I'm inclined to saying probably not without some larger customisations.

I can't say for sure, as I haven't tried those boards out (and I don't have access to any at the moment), but I'm guessing the port would involve changing repos, hardware settings, different firmwares, I'm not sure about the boot process, ...  again, it's really hard to say without additional research

diederikdehaas, 2015-12-01:
Thanks @andrewufrank :+1: 

As I don't have (access to) those boards, I don't know whether it works on other boards and/or what would be needed to get it to work on them. So I guess you'd just have to try and see.

Mausy5043, 2015-12-02:
I got a BeagleBone Black last week to do some hardware experimentations. And I noticed how completely different the install (requires flashing) and the boot processes (uses uBoot)  are compared to the Pi. I tried to work out in my mind what would need to be done to get something *like* `raspbian-ua-netinst` to work on the BBB. I'm inclined to believe that most hardware platforms will have significant differences from the Pi and each other.

I believe that it will be near impossible to create a netinstaller that is platform-independent or even multi-platform. I think a better approach would be to take the `raspbian-ua-netinst` concept and find a unique solution to implement that concept for each specific hardware platform.

N.B.: Most other platforms have an advantage over the Pi1 in that they can run Debian armv7 OOtB instead of depending on a man-in-the-middle organisation that supplies a working armv6 kernel.

diederikdehaas, 2015-12-02:
Thanks for the information @Mausy5043 :+1: 

Based on your U-Boot comment I also checked #240 and the [Debian **stretch** u-boot-sunxi package page](https://packages.debian.org/stretch/u-boot-sunxi) shows that it supports Cubieboard and Cubietruck and for odroid support you'd need the [u-boot-exynos package](https://packages.debian.org/stretch/u-boot-exynos).
That shows a number of issues:
1. It's from stretch/testing, while the installer is build from jessie/stable and looking at the 'list of files' it very much seems that odroid is only available on stretch.
2. It looks like the combined packages related to uboot are about 30MB.
3. The raspbian package seems to be stripped down to only contain the RPi bootloader

So all in all, _if_ it is possible, it will probably require a complete rewrite of the installer to support those other boards. It will also make the installer probably a LOT more complicated. As I also don't have those boards, I will not put time into it.
If others provide Pull Requests which would support those boards or make it easier to support those boards, I'm willing to consider them as long as they don't have a negative impact on the working for the RPi.

Mausy5043, 2015-12-02:
@diederikdehaas : Personally, I'd advise against integrating other platforms into this repo.

IMHO there should be a separate repo for each platform. I'm not suggesting you'd set those up. Other people (who are more familiar with those boards) should. I think it would be wiser to focus on developing for the RPi. 
I gather keeping everything working on just the RPi0, RPi1 and RP2 is challenging enough. :wink:

diederikdehaas, 2015-12-02:
I have no inclination to do it, but if others are, I'll judge it on its merits and according to the contribution guidelines.

mbt RPi0, ik heb het aan Sinterklaas gevraagd :wink:

Mausy5043, 2015-12-25:
Can we close this issue?

diederikdehaas, 2015-12-25:
I don't have a problem with that. @goranche what is your opinion?

Mausy5043, 2016-02-07:
Other boards are not going to be supported.
Closing.

closed 2016-02-07 by Mausy5043