debian-pi/raspbian-ua-netinst #305: Not working on all microSD sizes

aristosv, 2015-11-07:
I have a few 16GB microSD's and a couple of 512MB microSD's. This script is working on all my 16GB microSD's but on none of my 512MB microSD's.
Is there something that I must change to make it work on the 512MB microSD's?

Thanks.

goranche, 2015-11-08:
what does "doesn't work" mean? what happens, is there an error displayed, are there log files?

although I suspect a 512MB SD card might be too small to perform the install

aristosv, 2015-11-08:
Green and red leds are constantly on and the pi doesnt boot. Not even the rainbow is shown. Its no the pis fault because other sds work. 512MB is enough. Complete netinstall is smaller than 400MB

diederikdehaas, 2015-11-08:
> a 512MB SD card might be too small to perform the install

I'm quite sure that that statement is correct.
Even though the final installed size may be smaller the 512MB, it needs additional disk space during the installation.

> Is there something that I must change to make it work on the 512MB microSD's?

You _might_ get it to work on a 512MB card by using `preset=base` in your `installer-config.txt` or otherwise tweaking your `installer-config.txt` so that it can fit on a 512MB sd card. 
The question then is whether you still get a useful system...

diederikdehaas, 2015-12-01:
@aristosv What's the status of this issue? 

I don't think you can get `preset=server` to work on a 512MB card, but it may be possible with `preset=base` or `preset=minimal`.

closed 2015-12-02 by aristosv