debian-pi/raspbian-ua-netinst #308: Question about releases

bbinet, 2015-11-09:
As I can find any mailing list for the project, I ask my question here:

I noticed that there is two release branches on github: v1.0.x and v1.1.x but the latest release seem to be v1.0.7.
Is it safe to use the branch v1.1.x for a new project?
What are the states/differences of each branch? Do you plan to release v1.1.x soon?

Thanks.

diederikdehaas, 2015-11-09:
The difference between the v1.0.x and v1.1.x branch is that there are new configuration options available in the latter. So if someone develops a new feature which requires a new configuration option, that would have to be targeted at the v1.1.x branch.

The current plan is to release v1.0.8 in the not too distant future. I hope(d) that a new firmware version can be included so the installer also works on the Raspberry Pi Display. But it is taking longer then expected/hoped, so I may release v1.0.8 without it and release v1.0.9 with it when it becomes available. But that is likely the only change between v1.0.8 and v1.0.9.

When v1.0.8 is released, the focus will shift to the v1.1.x branch. There are still several features to be added and it needs to be tested thoroughly before v1.1.0 will be released. Once v1.1.0 is released, there can be no more configuration options added to that branch. See [Semantic Versioning](semver.org).
If you mean by 'soon' in the next couple of weeks, then the answer is no.
Is v.1.1x safe for a new project? AFAIK when you build an installer off the v1.1.x branch it should work and if not a bug report is much appreciated :wink: Whether it's safe for a new project, that's for you to decide as I don't know what that project is. But as v1.1.x is in development without a release, there may be significant changes implemented and that could impact your project.

bbinet, 2015-11-09:
Thanks for your informative answer.
Is there an issue to watch somewhere to know when the new rpi firmware version with rpi display support will be available?

diederikdehaas, 2015-11-09:
Yes, https://github.com/debian-pi/raspbian-ua-netinst/issues/282 and the upstream bug: https://bugs.launchpad.net/raspbian/+bug/1499046

closed 2015-11-09 by bbinet