debian-pi/raspbian-ua-netinst #312: 1.0.7 installs wheezy on a Pi B+

netdesk, 2015-11-19:
The README.md says:

> The format of the installer-config.txt file and the current defaults:
> ...
> release=jessie
> ...

I downloaded raspbian-ua-netinst-v1.0.7.img.xz, flashed it to the sd card using xzcat, inserted the card into the Raspberry Pi B+, booted and installed it, and got wheezy (it's using sysvinit and /var/log/raspbian-ua-netinst.log says release=wheezy). I didn't touch any files on the sd card prior to installing. I didn't even mount the card after the xzcat. Just xzcat, take out the card, insert in Pi, boot, wait, wheezy there.

Tried it twice, both times I got wheezy. Do I understand the README.md wrong?

diederikdehaas, 2015-11-19:
We probably shouldn't update files in the 'master' branch before making a release that matches with it.
So the [behavior is correct](https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.0.7/README.md) but understandable not what most ppl expect when reading the README from the master branch, which you see by default.

Currently (until the next release) the procedure should be, xzcat, mount the device again, create `installer-config.txt` with `release=jessie` as contents, unmount the device and plug it into the Pi.

netdesk, 2015-11-19:
Ok, thanks for clarification. I probably could have had a look at the branches and tags. As it's called unattended installer, I switched to no-brainer mode and just did what the README says (copy to SD and cross fingers). To be sure that I crossed fingers hard enough I did that twice o_0.

A solution may be, to keep only information in the README.md which stays the same during releases, and stuff the other parts into an INSTALL.md. Or name the release next to informations that change. So you wouldn't have to mess up your development and publishing workflow, just to keep us no-brainers from falling off the cliffs with the other lemmings.

closed 2015-11-19 by netdesk

diederikdehaas, 2015-11-19:
Nah, it's just something we should change. We already use the v1.0.x and v1.1.x branches for development and the merges into master don't add value and thus shouldn't be done before a release. It's only relevant for development of the installer and people who want to contribute already should know about the other branches (and target pull requests to those, not master).

But thanks for the sporty response :+1:

kpfleming, 2015-11-20:
You could also change the default branch of the repo to the 1.0.x branch, so that the default README is the one that matches the current releases.

diederikdehaas, 2015-11-20:
nope, that won't work. The problem was/is that we merged the changes from 1.0.x into master.
(development happens in the v1.0.x branch)