debian-pi/raspbian-ua-netinst #313: Move root to usb stick and keep boot on SD read only

Gusatomik, 2015-11-22:
Hello folks !

I just discover your raspbian-ua-netinst which seems to be great for self-host my personal-cloud.
But in a matter of life-ime, I would like to find a proper way to copy the root partition to an external USB storage.

I did it with ArchLinuxARM without troubles. But I'm not an expert and I'm stuck with your OS.
I use GParted to create an ext4 partition on my USB stick where I copy all the content of the second partition of the SD card (which contain the fresh raspbian install after the first boot). Then I modified the cmdline.txt to specify the new location of the root "root=/dev/sda2".

That was approcimately my process but I don't succed to get ssh after that...

Do you have any clue to proceed that way ?

Thank you in advance.

diederikdehaas, 2015-11-22:
If you put the installer on the sd card, you should mount it and create `installer-config.txt` with the following contents:
```
usbroot=1
release=jessie
```
The last setting is only needed when you want to install jessie directly, otherwise it will default to wheezy (for the v1.0.7 release). See https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.0.7/README.md#installer-customization for details.
This will install only `/boot/` on the sd card and the rest, ie `/` (root) on the usb drive. It will (re-)partition your usb drive, so make sure there aren't any files on it you'd like to keep.

HTH

Gusatomik, 2015-11-22:
Thank you very much diederikdehaas for your answer. After reading it I went back to the installer-config.txt example and shame on me the option was just in front of my eyes...

cheers

closed 2015-11-22 by diederikdehaas