debian-pi/raspbian-ua-netinst #317: even less software

aristosv, 2015-11-25:
Is there a way to configure this, so that it doesn't install additional software?
For example I don't want to install dbus, fake-hwclock, rsyslog and other software.
But I don't want to remove them after the installation. I don't even want them to get downloaded.
And I don't want to use the base install because it doesn't install network.

Can this be done?

diederikdehaas, 2015-11-25:
Use the `preset=base`, but add the packages you want to the `packages` config option.
As you can [see in `scripts/etc/init.d/rcS`](https://github.com/debian-pi/raspbian-ua-netinst/blob/master/scripts/etc/init.d/rcS#L324) the difference between `base` and `minimal` (and `server`) are a number of packages that get installed. But you can use `base` and add the packages you want to the `packages` option to fine tune it.
For ultimate control, there's even the `cdebootstrap_cmdline` config option.

aristosv, 2015-11-25:
Ok, I'll give it a try. Which package is the networking?

aristosv, 2015-11-25:
Nevermind ill find it thanks :)

closed 2015-11-27 by diederikdehaas