debian-pi/raspbian-ua-netinst #318: problem with installer-config.txt file - creating minimal install

aristosv, 2015-11-27:
I am trying to create a raspbian image with the minimum packages required. So I am using a base install and then using cdebootstrap_cmdline. Something like this:

#preset=base
#packages= # comma separated list of extra packages
#mirror=http://mirrordirector.raspbian.org/raspbian/
#release=jessie
hostname=raspberry
#domainname=
rootpw=raspberry
cdebootstrap_cmdline="--flavour=build --include=cpufrequtils,kmod,ifupdown,net-tools,ca-certificates,openssh-server"
bootsize=+64M
#rootsize=
#timeserver=time.nist.gov
#ip_addr=dhcp
#ip_netmask=0.0.0.0
#ip_broadcast=0.0.0.0
#ip_gateway=0.0.0.0
#ip_nameservers=
#online_config= # URL to extra config that will be executed after installer-config.txt
#usbroot= # set to 1 to install to first USB disk
#cmdline="dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 elevator=deadline"
#rootfstype=ext4
#rootfs_mkfs_options=
#rootfs_install_mount_options='noatime,data=writeback,nobarrier,noinit_itable'
#rootfs_mount_options='errors=remount-ro,noatime'

I must be doing something wrong though because after the reboot, I just get the rainbow and it gets stuck there. Do you know what I might be doing wrong in this installer-config.txt file?

diederikdehaas, 2015-11-27:
>Use the preset=base, but add the packages you want to the packages config option.
As you can see in scripts/etc/init.d/rcS the difference between base and minimal (and server) are a number of packages that get installed. But you can use base and add the packages you want to the packages option to fine tune it.

> For ultimate control, there's even the cdebootstrap_cmdline config option.

That was meant as one or the other. You should use `base` and the `packages` option.
Btw, you would be installing **wheezy** if you don't specify `release=jessie`.

I knew this was a horrible idea the moment [you asked](https://github.com/debian-pi/raspbian-ua-netinst/issues/317#issuecomment-159692272) which packet is networking. _Sigh_

aristosv, 2015-11-27:
just trying to learn. but since you mention it, which packages would install basic networking? would net-tools be enough?

diederikdehaas, 2015-11-27:
I'll ask you another question instead. Why do you want to remove `fake-hwclock` and what consequences do you think that'll have?

aristosv, 2015-11-27:
I want to remove all of these

kmod
fake-hwclock
ntp
vim-tiny
wget
rsyslog
dialog
locales
less
man-db

and install busybox-syslogd and ntpdate

Purpose being, I want to create a read-only raspbian image. So I need to remove as much as possible and run only the absolute necessary.

diederikdehaas, 2015-11-27:
I know you want to remove even more packages, but that doesn't answer my question.
I have a **strong** feeling you want to remove package for the wrong reasons without understanding the consequences of it. 
That you haven't been able to figure which package(s) are responsible for networking, enforces that feeling.
> I want to create a read-only raspbian image. So I need to remove as much as possible and run only the absolute necessary.

Your conclusion is wrong. You can have a read-only system with (a lot) more packages installed then the `server` preset installs, which is already contains a pretty minimal amount of packages.

aristosv, 2015-11-27:
I understand that a fully fledged operating system can be read-only. But in the case of running raspbian off a micro-sd, I need it to be as small as possible. The server preset, includes fake-hwclock, rsyslog and other packages that are not necessary for the purpose I want to build this image.

I just need it to be as "empty" as possible, so I can install only what I need. Ideally it would initially only be running openssh-server and apt-get. (with network connectivity of course)

aristosv, 2015-11-28:
Like you suggested I chose the base installation and installed the packages I wanted. The installer-config.txt was like this:

preset=base
packages=ifupdown,net-tools,openssh-server
release=jessie
hostname=raspbian
rootpw=raspbian
bootsize=+64M

The installation was completed successfully, but when it rebooted it got stuck at the rainbow image.
Any idea why?

diederikdehaas, 2015-11-28:
You probably ran into https://github.com/debian-pi/raspbian-ua-netinst/issues/253

diederikdehaas, 2015-12-01:
Version 1.0.8 of the installer has been released, so the rainbow problem should now be solved.

closed 2015-12-02 by aristosv