debian-pi/raspbian-ua-netinst #319: Will there be a release?

redhead, 2015-11-27:
Hi, I would like to ask if a release is going to be soon? I would really like to use the wi-fi support. I wanted to build the image myself, but I can't (not a linux user and no so much space on sd-card for raspberry too). There was no release since May :(

diederikdehaas, 2015-11-27:
There will be a release soon, but it'll be based on the v1.0.x branch, thus without the wifi support. 
A release based on v1.1.x (with wifi) is not coming soon, so you should try to find a way to build it yourself. Maybe setup a linux (Debian) system in a virtual machine like VirtualBox?

redhead, 2015-11-27:
I see, thanks. Should I build the master or the 1.1.x branch then? Which one is "more stable"?

diederikdehaas, 2015-11-27:
The master branch and v1.0.x are (essentially) the same (now) and are the most stable ones. 
That the last release was from May doesn't mean it's outdated as it retrieves and installs the latest packages available at the moment of the installation itself, not the packages from May.
The primary difference between v1.0.x/master and release v1.0.7 is switching the default release to jessie, but you can force the installation of jessie also with v1.0.7 by specifying `release=jessie` in `installer-config.txt`.

But if you want to install over wifi, you **need** the v1.1.x branch.

redhead, 2015-11-27:
> But if you want to install over wifi, you need the v1.1.x branch.

Right, thanks.

redhead, 2015-11-27:
Oh, could you please tell me where to put the drivers for the wi-fi adapter? Do I have to do it before building?

diederikdehaas, 2015-11-27:
The drivers for your wifi adapter are not present in the kernel? Then it could become challenging :wink: 
You can make your own driver archive and unpack that by placing a script in `installer-config.txt`:
```
if [ -f /bootfs/config/drivers.tar.gz ] ; then
  echo -n "Loading drivers... "
  tar -zxf /bootfs/config/drivers.tar.gz -C /
  echo "OK"
  echo -n "Running depmod to update the list of modules... "
  /sbin/depmod -a
  echo "OK"
fi
```
And in `drivers.tar.gz` you create a structure `/lib/modules/3.18.0-trunk-rpi2/kernel/drivers/net/wireless/` for a Pi2 or `/lib/modules/3.18.0-trunk-rpi/kernel/drivers/net/wireless/` for a Pi1 and place your drivers in there. 
And you also have to make sure that the drivers are available in the target system, probably by scripting that in `post-install.txt`, otherwise your wifi won't work when the system reboots.

Now you know _a_ reason why v1.1.x won't be released just yet :stuck_out_tongue_winking_eye:

redhead, 2015-11-27:
Well, actually I don't know for sure the driver is not present in the kernel.. before I followed [this way](https://www.raspberrypi.org/forums/viewtopic.php?p=462982#p462982) for the TL-WN725N wifi adapter, and it worked when I did this manually. So I thought I have to do this in the net installer too. 

Or do you know what driver should I use that is present in the kernel so it works? I have no idea.. I tried to find "8188" (that is used in that tutorial above) and it found some but not in the .../net/wireless folder. It was folder `/drivers/staging/rtl8188eu/r8188.ko`.

diederikdehaas, 2015-11-27:
Welcome to the kernel/driver hell created by the Raspberry Pi Foundation :stuck_out_tongue_closed_eyes: 

MrEngman provides a HUGE service for users of the Pi by providing drivers for every tiny kernel/firmware release that the RPF has. I call those kernel raspberrypi.org kernels.
This installer uses what I call a raspbian.org kernel and the precise reason is that it doesn't create the mess the RPF still creates and it also provides kernel-header packages which make it quite easy to compile your own kernel if you should need that.
The kernel that gets installed and used (by default) on the installed system is also a raspbian.org kernel. 
The key and critical point is that the kernel and the kernel module (`r8188eu.ko` in your case) need to match exactly and the kernel modules (=drivers) provided by MrEngman don't match the one from the installer/installed system, so you can't use those.

> found some but not in the .../net/wireless folder. It was folder /drivers/staging/rtl8188eu/r8188.ko

And that makes you VERY lucky, because that driver is (also) present in the raspbian.org kernel  :wink: 

You need to download http://archive.raspbian.org/raspbian/pool/main/l/linux-3.18/linux-image-3.18.0-trunk-rpi2_3.18.5-1~exp1%2brpi19_armhf.deb for the Pi2 or http://archive.raspbian.org/raspbian/pool/main/l/linux-3.18/linux-image-3.18.0-trunk-rpi_3.18.5-1~exp1%2brpi19_armhf.deb for the Pi 1 and extract the `r8188.ko` file from that archive and put it in the archive you need to create yourself (see my previous reply). 
Whether you put it in `/lib/modules/3.18.0-trunk-rpi/kernel/drivers/net/wireless/` or `/lib/modules/3.18.0-trunk-rpi/kernel/drivers/drivers/staging/` probably doesn't matter.
With that it will probably work (don't forget to put it in `drivers_to_load` parameter).

If your installed system doesn't load the driver automatically, then you should put it in `/etc/modules` to force it's loading during boot up.

redhead, 2015-11-28:
Why do I need to download it extra when it is already present in the kernel? I found it in the tmp folder (before it is deleted)... I am a bit comfused... :/

Thanks for help though, it must take a lot of time to help noobs like me :) ;)

diederikdehaas, 2015-11-28:
> Why do I need to download it extra when it is already present in the kernel?

It is present in the full kernel which does get installed on the 'target' system (ie which is used when you reboot).
I _thought_ there were no wireless drivers present at all in the kernel which is used during installation ... until I realized I was looking in the wrong branch ...
And then I found/realized that the solution for you can be made MUCH simpler :smile: 

At https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.1.x/build.sh#L497 you can see that the `kernel/drivers/net/wireless` directory is included in full ... so if you add the following line to your local build.sh (after line 497), and then [rebuild the installer](https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.1.x/BUILD.md) (./clean.sh ; ./build.sh ; sudo ./buitdroot.sh) then it will be included in the installer kernel and you don't have to go through all the hoops:
```
cp -r tmp/lib/modules/$KERNEL_VERSION/kernel/drivers/staging/rtl8188eu rootfs/lib/modules/$KERNEL_VERSION/kernel/drivers/net/
```
(but you still have to specify it in `drivers_to_load` :wink: )

> it must take a lot of time to help noobs like me

It does, but it's often also useful for me, like finding the above thing, which will be added to the repo in a future commit :+1: 
And I like helping people (as long as they put an effort in it themselves too).

redhead, 2015-11-28:
Nice! By the way, I am not at all familiar with linux kernels and drivers, so it makes me think why did they do it manually in the tutorial if it is already present in the kernel. Is it 100% THE driver I am looking for? Can I rely only by the name 8188eu that they used too?

redhead, 2015-11-28:
Oh, I was googling and found that they maybe different - there is `8188eu.ko` (used in the tutorial) and `r8188eu.ko` (in kernel), notice `r` in the beginning.

diederikdehaas, 2015-11-28:
> why did they do it manually in the tutorial if it is already present in the kernel. 

The 'staging' area is the place where 'things' (like drivers) are which are not quite ready to be included in the mainline kernel. And that alone is a good reason not to use it.
In its [kernel configuration file](https://github.com/torvalds/linux/blob/8e483ed1342a4ea45b70f0f33ac54eff7a33d918/drivers/staging/rtl8188eu/Kconfig), if you look at the _select_ lines you can see `WIRELESS_EXT` which is the **OLD** wireless extension. 
In [drivers which are part of mainline](https://github.com/torvalds/linux/blob/master/drivers/net/wireless/realtek/rtl8xxxu/Kconfig) you can see that they have `MAC80211` and that's the 'NEW' wireless extensions.
I don't know, but I _think_ MrEngman found an alternative driver source which does use the new wireless extensions.

> they maybe different - there is 8188eu.ko (used in the tutorial) and r8188eu.ko (in kernel)

If you use the drivers from staging, I think you should use `drivers_to_load=r8188eu`. See the kernel config file I linked earlier.

> Is it 100% THE driver I am looking for? 

Not sure, but it's worth a try as it beats the alternative (compiling the kernel module yourself and performing all the steps I explained at the beginning of this issue).

diederikdehaas, 2015-12-04:
If you still have questions regarding this issue, feel free to continue to use this issue, but I'm 'merging' this with issue #52 as that (already) is about additional drivers.

closed 2015-12-04 by diederikdehaas