debian-pi/raspbian-ua-netinst #321: How to upgrade from Wheezy to Jessie?

ffes, 2015-11-30:
I have a rpi1 as simple server at home for over a year now (used v1.0.5 back then to install). It runs fine and I have no intend to reinstall everything.
But for some of the packages I need the new version that is included in Jessie.

I can't find any reference about "How to upgrade from Wheezy to Jessie" or did I overlook it?

Do I just replace `wheezy` with `jessie` in `/etc/apt/sources.list`?
If so, are there any special things I need to think about (apart from a decent backup)?

diederikdehaas, 2015-11-30:
> I can't find any reference about "How to upgrade from Wheezy to Jessie" or did I overlook it?

There isn't and there will be no 'guide' to upgrade from Wheezy to Jessie because it has nothing to do with this project.
The standard way to upgrade a Debian and thus Raspbian system is replacing `wheezy` with `jessie` in `sources.list` and possibly in files in `/etc/apt/sources.list.d/` if applicable and then doing an `apt-get dist-upgrade`

JedMeister, 2015-11-30:
There may be some RPi idiosyncrasies that I am unaware of (I don't play with my Pi anywhere near enough) but the Debain docs are a good place to start. Might be worth checking on the RPi forums too? 

This page is for i386 but the theory remains the same:
https://www.debian.org/releases/stable/i386/release-notes/ch-upgrading.html

diederikdehaas, 2015-12-25:
It really should be as simple as I described earlier and the chances of a smooth upgrade are rather good (I never had any problems).
This [thread on the raspberrypi forums](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=121880) could also help if you had installed various packages from the raspberrypi archives.

I'm closing this issue now as the OP hasn't responded since opening the issue.
Feel free to reopen it if needed.

closed 2015-12-25 by diederikdehaas