debian-pi/raspbian-ua-netinst #322: got the rainbow again

aristosv, 2015-12-03:
I am using this configuration file
hostname=raspbian
rootpw=raspbian
bootsize=+64M
cdebootstrap_cmdline='--flavour=minimal --include=kmod,ifupdown,net-tools,isc-dhcp-client,openssh-server,iputils-ping,wget,ca-certificates,dialog,less'

After the first reboot I got the rainbow. So I modified it like this.

hostname=raspbian
rootpw=raspbian
bootsize=+64M
packages=${kernel_meta_package}
cdebootstrap_cmdline='--flavour=minimal --include=kmod,ifupdown,net-tools,isc-dhcp-client,openssh-server,iputils-ping,wget,ca-certificates,dialog,less'

But I got the rainbow again.
What I'm I doing wrong?
Thanks

Mausy5043, 2015-12-03:
Instead of using the `cdebootstrap` you might prefer to use the entry `preset=server`.

diederikdehaas, 2015-12-03:
which version of the installer did you use?
And you should use either `preset` and `packages` or `cdebootstrap_cmdline` but not a combination of both.

aristosv, 2015-12-03:
@Mausy5043
I am using cdebootstrap because I need the additional customization.

@diederikdehaas
I am using version 1.8.
I understand what you're saying. Can I add ${kernel_meta_package} in cdebootstrap_cmdline?
I though this problem was fixed in version 1.8. Could it be something else?

diederikdehaas, 2015-12-03:
You should have a `/var/log/raspbian-ua-netinst.log` file. Can you share it?
One of the things that got fixed with 1.0.8 is that a kernel is always installed when using a **preset**.

I think that you can use `kernel_meta_package`

aristosv, 2015-12-03:
I added kernel_meta_package and tried again, but got the same result. Here's the log file: http://pastie.org/10603201

There's a message near the end that says:

Kernel version: ''
Copying kernel to /boot/kernel7.img... cp: can't stat '/rootfs/boot/vmlinuz-': No such file or directory
FAILED !

Mausy5043, 2015-12-03:
Hmm: `W: Can't find package kernel_meta_package for inclusion`

aristosv, 2015-12-03:
Same behavior occurs with, or without the kernel_meta_package

Mausy5043, 2015-12-03:
Can you see a `vmlinuz-*` file on the SD-card? 
If yes: What is the exact name?
else: :fearful:

Mausy5043, 2015-12-03:
I have a feeling something went wrong in the `make build`. Do you still have the output of that, or could you rebuild the image and post the output?

diederikdehaas, 2015-12-03:
It looks like you didn't include `${kernel_meta_package}`, but just `kernel_meta_package` and that's not a valid package name. The former is a variable reference which should be replaced with `linux-image-rpi-rpfv` for a Pi 1 (or 0) or `linux-image-rpi2-rpfv` for a Pi 2.
Maybe it's better/easier to use the applicable value in your `--include` list ...

diederikdehaas, 2015-12-03:
@aristosv Couldn't you just have re-opened issue #318 ? 
This is just a continuation of that same issue ...

aristosv, 2015-12-04:
I would but I was under the impression it was fixed with the latest update, so I though it may be something else.

This did the job

hostname=raspbian
rootpw=raspbian
bootsize=+64M
cdebootstrap_cmdline='--flavour=minimal --include=linux-image-rpi2-rpfv,kmod,ifupdown,net-tools,isc-dhcp-client,openssh-server,iputils-ping,wget,ca-certificates'

and this is the log file

http://pastie.org/10609315

closed 2015-12-04 by aristosv

diederikdehaas, 2015-12-04:
great :+1: