debian-pi/raspbian-ua-netinst #326: Pi Zero doesn't boot

barneyman, 2015-12-10:
The latest 108 release does not work with the zero - it just sits there flashing a green LED at me - no HDMI boot

Confirmed the Zero is operational by booting from a (netinst generated) Pi2 rasbian image

goranche, 2015-12-10:
I'd love to test it, but haven't received my Zero yet... :unamused:

diederikdehaas, 2015-12-10:
I don't have a Zero either, so I can't check it, but I believe that new firmware is needed. For [another issue](https://github.com/debian-pi/raspbian-ua-netinst/issues/282) I had already [requested it](https://bugs.launchpad.net/raspbian/+bug/1499046), but as of yet haven't received a response.

Mausy5043, 2015-12-10:
How did you achieve an internet connection?
The Pi0 is in essence a Pi1 without ethernet and with miniUSB and miniHDMI connectors. If you're trying to boot a Pi2 image that might be a cause.

barneyman, 2015-12-10:
the pi2 image is a working wheezy, built via netinst, on a real pi2, some months back - sorry for the confusion

diederikdehaas, 2015-12-10:
@barneyman do you have the `raspberrypi-bootloader` package installed and/or have you run `rpi-update` (recently)? 
Can you 'unzip' a `.tar.xz` archive?

barneyman, 2015-12-10:
(on my other pi1s, pi2s)
1.  no to '<raspberrypi-bootloader>' package, 
2. yes to '<rpi-update>', 
3. yes to '<.tar.xz>'

would you like me to try something?

diederikdehaas, 2015-12-10:
The image likely booted due to `rpi-update`.
> would you like me to try something?

Yes. http://cknow.org/rpi/firmware-fc6c9896.tar.xz contains a more recent set of firmware files. What I want you to do it put the v1.0.8 release on a sd card like you normally do. But before you put it in the Pi Zero, I like you to extract the `.tar.xz` file onto it, which will override the firmware files with newer versions. And I'd like to know if that _would_ make it start up.

barneyman, 2015-12-10:
ok - will do - please give me an hour or two ..

barneyman, 2015-12-10:
1. Downloaded 1.08
2. Formatted SD card (2gb) to FAT
3. Exploded 1.08 onto card
4. Exploded 'firmware...xz' onto card
5. booted zero

no change ... still doesn't boot - no HDMI start up - sorry - i don't have an FTDI board here, or i'd snoop the tty

diederikdehaas, 2015-12-10:
Thanks for trying :+1: 

Then my guess is that also a kernel update is needed to get it working on the Pi Zero.

barneyman, 2015-12-10:
ok NP - if you'd like to try anything else, ask away ...

BenjaminHae, 2016-08-20:
Did someone get it to work?
I tried the latest v1.1.x branch. Screen stops at rainbow and the activity led flashes 7 times then pauses and repeats the flashing.

diederikdehaas, 2016-08-20:
I (still) don't have one and I'm not sure I'll get one either.
But 7 flashes indicates that it can't find kernel.img so maybe it isn't processing `config.txt`.
What if you'd add a `[pi0]` section to it with the same contents as the `[pi1]` section?

BenjaminHae, 2016-08-21:
It worked. Is there anything to test before I'm opening a PR?

Mausy5043, 2016-08-21:
@BenjaminHae : If the Pi is running now, what you're showing there is probably an old log.

BenjaminHae, 2016-08-21:
@Mausy5043 I'm pretty sure it's the log of the now working installation.
All the other times I booted the pi I only got to the rainbow screen so no log should have been created- Even if one had been created it should not have gotten to the removal of cdebootstrap-...

Mausy5043, 2016-08-21:
It's just that the line 
> You have 10 seconds to hit ENTER to get a shell..

is output just before this piece of code is executed:
```
    read -t 10 || reboot && exit
    sh
```

It is very unlikely that you would end up with an installed system when you see that text.

BenjaminHae, 2016-08-21:
Yeah, but

> Removing cdebootstrap-helper-rc.d... 

occurs in line 1.162 of rcS. The only things happening afterwards are saving the current time, copying the logfile to `/rootfs/var/log` and unmounting of `/rootfs` and `/rootfs/boot`.
So I think having a working system after what happened in the logs is very likely.

diederikdehaas, 2016-08-22:
I think we're close to also having Pi0 support
> So I think having a working system after what happened in the logs is very likely.

And that's probably not the case, although it should be easy to fix.
It (very much) looks like you now have a `[pi0]` section in `config.txt` of the **installer**, but as there is no line which will also write it to `config.txt` of the **installed** system, you'd likely get a rainbow screen.
Can you share your installer log file? (Remember to blank out passwords etc)

BenjaminHae, 2016-08-22:
You are right I have the `[pi0]` section in the installer's `config.txt`, it is missing on the installed system, but booting isn't a problem. It seems to be a stable system, but with cdebootstrap-helper-rc.d still installed.

I only kept the logs from when I used  [goranche's read-only installation](https://github.com/goranche/raspbian-ua-netinst-ro) with a modified `post-install.txt`. This is the log: [raspbian-ua-netinst-20160821T084356.log](https://gist.github.com/BenjaminHae/969e9e5d12352decbd1b20900dc81659)

diederikdehaas, 2016-08-22:
The error which caused `fail()` to be called does seem related to the read-only installation.
> libkmod: ERROR ../libkmod/libkmod-module.c:1903 kmod_module_get_holders: 
could not open '/sys/module/8192cu/holders': No such file or directory

But that may be caused by sth else ...

I'm not sure whether the failure wrt `cdebootstrap-helper-rc.d` is caused by the read-only installation or that it is caused by sth else.
What you could do is doing a 'plain' installation, with the latest from our `v1.1.x`, and see how that fares. Sharing the installation log should be helpful.

BenjaminHae, 2016-08-22:
Ok, now the installation went without a problem: [raspbian-ua-netinst.log](https://gist.github.com/BenjaminHae/3ce38857fa2d1e0b4e72796bbaed4ecf)

The installer `config.txt` contained the `[pi0]` section. The install**ed** `config.txt` doesn't. The system boots anyhow.

diederikdehaas, 2016-08-22:
> The installer config.txt contained the [pi0] section. The installed config.txt doesn't. The system boots anyhow.

That is likely due to `Copying kernel to /boot/kernel.img... OK`
Looks to me that adding Pi Zero support is a piece of cake :-D

diederikdehaas, 2016-08-23:
With the merge of #421 I think we can close this issue. 
If you feel otherwise, please reopen it and specify what you did and what isn't working.

PR #421 (2016-08-22): pi zero support
fixes #326
closed 2016-08-23 by diederikdehaas