debian-pi/raspbian-ua-netinst #329: Can't see any checksums

rustleg, 2015-12-11:
I have searched for any SHA1 or MD5 checksums for your downloads but can't find any. I expect it's my ineptitude, so if you have provided them somewhere please point me to them.

diederikdehaas, 2015-12-11:
Good point :+1: 

They're not there, so it's not your fault, but ours.

diederikdehaas, 2015-12-11:
Here are the SHA1 values:
```
3ff77fffda84400a033d160c1adf6078802f929b  raspbian-ua-netinst-v1.0.8.img.bz2                                                                     
d17dae9d629d7b1b97c4894d9a6eeb33978f1ae9  raspbian-ua-netinst-v1.0.8.img.xz                                                                      
c512ad7246f0568446d58e1bedf20d5daece0081  raspbian-ua-netinst-v1.0.8.zip
```

goranche, 2015-12-12:
@rustleg are you ok with closing this issue now?

rustleg, 2015-12-12:
Ah yes, I see now that you close posts, sorry.

Thanks diederikdehaas for posting the checksums and even more thanks for all your work on this project - very much appreciated.

closed 2015-12-12 by rustleg

diederikdehaas, 2015-12-12:
Cheers :+1: