debian-pi/raspbian-ua-netinst #334: Rainbow on boot after installing of fuse(sshfs?)

BenjaminHae, 2015-12-14:
I'm not sure if this belongs here or somewhere more general, but I did use this installer(not a release, just compiled the master branch on commit https://github.com/debian-pi/raspbian-ua-netinst/tree/2763c86dbe14f8090cfa85262f7eb1e7a31bfd30 -- not sure how to reference this right, sorry).

I installed `sshfs` a few days later and got this message

    update-initramfs: deferring update (trigger activated)
    Setting up sshfs (2.5-1) ...
    Processing triggers for libc-bin (2.19-18+deb8u1) ...
    Processing triggers for initramfs-tools (0.120) ...
    ln: failed to create hard link ‘/boot/initrd.img-3.18.0-trunk-rpi.dpkg-bak’ => ‘/boot/initrd.img-3.18.0-trunk-rpi’: Operation not permitted
    update-initramfs: Generating /boot/initrd.img-3.18.0-trunk-rpi

The `ln` error didn't surprise me.
But after a restart I got the rainbow screen on boot. I fixed it by copying an old `/boot/initrd.img-3.18.0-trunk-rpi` back to `/boot` as this was the only file with a change date on the day of the installation of fuse. Everything seems to work fine now.

Maybe you know what happened?

P.S: Thank you very much for your work. I really appreciate it!
  If this issue does not belong here: sorry I didn't know any better. Please point me to another direction.

diederikdehaas, 2015-12-14:
Can you post the contents of `/boot/config.txt` ?

BenjaminHae, 2015-12-14:
````
[pi1]
kernel=vmlinuz-3.18.0-trunk-rpi
initramfs initrd.img-3.18.0-trunk-rpi followkernel
dtparam=spi=on
gpu_mem=16
boot_delay=1
````
Pi B `/boot` is on a 1Gb SD card, `/` is on a usb flash drive. with `/home` on a separate partition. If that helps.

diederikdehaas, 2015-12-14:
I'm guessing you have a Pi 1B+ and there are issues with DT in combination with initramfs with that model. See https://github.com/raspberrypi/linux/issues/914 for more on that.  
Does it boot if you comment out the `initramfs` line?

diederikdehaas, 2015-12-25:
Ping?

Alternatively you could try to re-generate the initramfs as follows and that may resolve the issue too (although possibly only temporary): `/etc/kernel/postinst.d/initramfs-tools $(uname -r)`

BenjaminHae, 2015-12-25:
It may take a while until I have time to try this as I have to finish my masters thesis until the end of january.

But I will write as soon as I have tried your tips.

diederikdehaas, 2015-12-25:
Thanks and good luck :+1:

Mausy5043, 2016-03-20:
@BenjaminHae : Love to hear if your problem is solved

BenjaminHae, 2016-03-20:
Oh. I forgot about this and I don't think I have the original files anymore. But I'm pretty sure it was because of device tree. Sorry

closed 2016-03-20 by BenjaminHae