debian-pi/raspbian-ua-netinst #336: proper use of post‑install.txt

aristosv, 2015-12-16:
This is not an issue, its more of a question. I've been having problems with some of the commands I'm using in post‑install.txt, so I wanted to know what is the proper way to use post‑install.txt.

Is there a specific formatting I must follow?

For example, this works:
if [ -f /rootfs/lib/udev/write_net_rules ] ; then
  echo -n "Modifying /lib/udev/write_net_rules so net rules don't get written ..."
  sed -i 's/\(write_rule "\$match" "\$INTERFACE" "\$COMMENT"\)/#\1/' /rootfs/lib/udev/write_net_rules
  echo "OK"
fi

but this does not:
echo -n "Creating Folders..."
  mkdir {/rootfs/root/media,/rootfs/root/config,/rootfs/root/onboot}
echo "OK"

Thanks

diederikdehaas, 2015-12-16:
> Is there a specific formatting I must follow?

No, no specific formatting. But it is executed by the busybox shell. So if you install busybox on your local system and put `#!/bin/busybox sh` at the top of your script file, you should be able to 'debug' your scripts outside of an installation on the Pi.

kpfleming, 2015-12-16:
Also, there's no need to wrap multiple arguments to ```mkdir``` in curly-braces. It accepts multiple arguments directly, they can just be specified on the command line separated by spaces. Curly braces are used when you want to create an expansion.

closed 2015-12-16 by aristosv