debian-pi/raspbian-ua-netinst #337: Unmet dependencies error during installation

aristosv, 2015-12-16:
I get a message saying "Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution)." during installation.

This is my installer-config.txt
packages=ssmtp,mailutils,lftp,mpg123,sshpass,bc,nano,psmisc,alsa-utils,curl
hostname=ClientX
rootpw=passhere
bootsize=+64M

This is the full installation log.
http://pastie.org/10636612

diederikdehaas, 2015-12-16:
What happened after you did `apt-get -f install` ?

aristosv, 2015-12-16:
I never managed to press enter to get the shell, because it rebooted.

aristosv, 2015-12-16:
And restarted the installation.

aristosv, 2015-12-17:
Could it be something with the repository? I left a Pi running through the night and I got 12 log files on the microSD card. The ones with the unmet dependencies error, always fail at the same package. 

P: Configuring helper cdebootstrap-helper-apt
E
: 
Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).

E: Internal error: install

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...

Some of other log files, had the following messages.

P: Retrieving exim4-config
wget: can't connect to remote host (5.153.225.207): Connection timed out
E: Couldn't download pool/main/e/exim4/exim4-config_4.84-8_all.deb!

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...

P: Retrieving init
wget: error getting response: Connection reset by peer
E: Couldn't download pool/main/i/init-system-helpers/init_1.22_armhf.deb!

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...

aristosv, 2015-12-17:
So I started a new setup, got the same error message, had a keyboard already hooked up, got to the console and run "apt-get -f install". I got a message saying "apt-get: not found"

diederikdehaas, 2015-12-17:
> Could it be something with the repository?
wget: can't connect to remote host (5.153.225.207): Connection timed out

It very much looks like that is the case, because that IP address corresponds to `mirrordirector.raspbian.org`. Can you try it with a different mirror? 
Pick one that is close by for you from http://raspbian.org/RaspbianMirrors

aristosv, 2015-12-17:
I tried the first one (http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian) and downloading was actually much faster. But at the end I got this error again.
And for some reason its always at the same point. During Configuring helper cdebootstrap-helper-apt

P: Configuring helper cdebootstrap-helper-apt
E
: 
Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).
E: Internal error: install
Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...

aristosv, 2015-12-17:
ok here's another discovery. I removed the "packages" line from installer-config.txt and restarted the installation. This is what I got when it reached "P: Configuring helper cdebootstrap-helper-apt"

P: Configuring helper cdebootstrap-helper-apt
debconf: delaying package configuration, since apt-utils is not installed
E
:
Can not write log (ls /dev/pts mounted?) - posix_openpt (Z: No such file or directory)

I really don't think this is a repository issue. I tried different repositories and the result was the same. It seems to be something with apt-utils.

Any ideas?

JedMeister, 2015-12-17:
I suggest that you do test with a standard vanilla (known good) installer. I.e. don't customise it at all. Then at least you'll have a better point to start troubleshooting from.

diederikdehaas, 2015-12-17:
> P: Configuring helper cdebootstrap-helper-apt
debconf: delaying package configuration, since apt-utils is not installed
E: Can not write log (ls /dev/pts mounted?) - posix_openpt (Z: No such file or directory)

That is normal and expected behavior. 
So it looks like it's something from your package selection.

aristosv, 2015-12-18:
It seems ssmtp is causing the issue. Any idea why a package would screw up the installation so badly?

update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
P: Configuring package fake-hwclock
hostname: Name or service not known
dpkg: error processing package ssmtp (--configure):
 subprocess installed post-installation script returned error exit status 1
ln: 
failed to create hard link '/boot/initrd.img-3.18.0-trunk-rpi2.dpkg-bak' => '/boot/initrd.img-3.18.0-trunk-rpi2'
: Operation not permitted
Errors were encountered while processing:
 ssmtp
E: Sub-process /usr/bin/dpkg returned an error code (1)
E: Internal error: install

JedMeister, 2015-12-18:
@aristosv 
>hostname: Name or service not known
Yep looks like a known issue when installing ssmtp if you don't have an FQDN configured properly...
I.e. you need to configure /etc/hosts and /etc/hostname before you install ssmtp.

Although I'm not quite sure how it would be best resolved...

Have you tried installing ssmtp on a vanilla install (after firstboot but before you do anything else)? At least then you'll know the requirements for install in this environment. Maybe the hostname config by default is inadequate for ssmtp? Or perhaps it's just a timing issue? 

Once you've isolated what the actual problem is, then you can work out the best way to fix it! :smile:

aristosv, 2015-12-18:
Maybe I should try installing ssmtp using "post-install.txt". I have doubts though..

aristosv, 2015-12-18:
So from what I'm reading, it seems that ssmtp fails because at the time of its installation, the host name in /etc/hosts is not yet set, or the entries in /etc/hosts and /etc/hostname don't match.

I don't know if that's true, but people say that as soon as they set the host name, ssmtp installation works. So theoretically, I could set the hostname and install ssmtp in post-install.txt.

But the real question is, at which point of the setup is the hostname set? Before, or after the package installation?

diederikdehaas, 2015-12-18:
> ssmtp fails because at the time of its installation, the host name in /etc/hosts is not yet set

That is correct. The hostname is set at https://github.com/debian-pi/raspbian-ua-netinst/blob/master/scripts/etc/init.d/rcS#L528 while the package installation happens https://github.com/debian-pi/raspbian-ua-netinst/blob/master/scripts/etc/init.d/rcS#L506.
Installing `ssmtp` in `post-install.txt` should fix that.

I'll think about if we should move that (and possibly other) settings earlier in the installation process.

aristosv, 2015-12-18:
I tried installing ssmtp using the following:

echo -n "Installing SSMTP..."
  /rootfs/usr/bin/apt-get install -y ssmtp
echo "OK"

But the log file reports:

Installing SSMTP.../rootfs/usr/bin/apt-get: error while loading shared libraries: libapt-pkg.so.4.12: cannot open shared object file: No such file or directory

Any idea what could it be now?

diederikdehaas, 2015-12-18:
> /rootfs/usr/bin/apt-get install -y ssmtp

Replace that with `chroot /rootfs /usr/bin/apt-get -y install ssmtp`

aristosv, 2015-12-19:
Great! That worked! Thanks.

echo -n "Setting Hostname..."
  hostname ClientX
echo "OK"

echo -n "Installing SSMTP..."
  chroot /rootfs /usr/bin/apt-get -y install ssmtp
echo "OK"

closed 2015-12-19 by aristosv

Mausy5043, 2015-12-19:
@aristosv : Tip: https://guides.github.com/features/mastering-markdown/ :wink: