debian-pi/raspbian-ua-netinst #338: Emailing the log file

aristosv, 2015-12-19:
I have set everything up in post-install.txt to be able to sent email, and I have this command in there also.
```
echo -n "Sending Logfile..."
  /rootfs/usr/bin/mail -s "$HOSTNAME Setup Completed" aristos@aristos.net < /rootfs/var/log/raspbian-ua-netinst.log
echo "OK"
```

But in the logfile I get this
```
Sending Logfile.../etc/init.d/rcS: /bootfs/post-install.txt: line 78: can't open /rootfs/var/log/raspbian-ua-netinst.log: no such file
OK
```

Is the log file written somewhere else, and copied to /var/log after the installation finishes?

Mausy5043, 2015-12-19:
yep
https://github.com/debian-pi/raspbian-ua-netinst/blob/master/scripts/etc/init.d/rcS#L3  says:
```
LOGFILE=/tmp/raspbian-ua-netinst.log
```
And then at the end:
https://github.com/debian-pi/raspbian-ua-netinst/blob/master/scripts/etc/init.d/rcS#L690-L691

```
# copy logfile to standard log directory
cp $LOGFILE /rootfs/var/log/raspbian-ua-netinst.log
```

aristosv, 2015-12-19:
I tried this
```
/rootfs/usr/bin/mail -s "$HOSTNAME Setup Logfile" my@email.net < /rootfs/tmp/raspbian-ua-netinst.log
```
But I got a message saying 
```
can't open /rootfs/tmp/raspbian-ua-netinst.log: no such file
```

I can't use /rootfs/var/log/raspbian-ua-netinst.log because the "send email" command runs before the copying of the log file in /rootfs/var/log

diederikdehaas, 2015-12-19:
> can't open /rootfs/tmp/raspbian-ua-netinst.log: no such file

That's because it is `/tmp/raspbian-ua-netinst.log`.

@Mausy5043 was kind enough to point the solution out for you ( :+1: ), but IMO you could've easily found it yourself. The whole installation process is coded in https://github.com/debian-pi/raspbian-ua-netinst/blob/master/scripts/etc/init.d/rcS. If there are parts of the installation script that you don't understand, I'm happy to explain it to you and/or improve the documentation, but I do expect you to put a bit more effort into it yourself then you've shown here.

aristosv, 2015-12-19:
Ok. Thanks.

closed 2015-12-19 by aristosv