debian-pi/raspbian-ua-netinst #349: wpa_supplicant.conf is stored at an odd location

diederikdehaas, 2015-12-29:
Currently `scripts/etc/init.d/rcS` has the following code block (line 776-778):
```
if [ "$ifname" != "eth0" ]; then
    echo "wpa-conf /boot/config/wpa_supplicant.conf" >> /rootfs/etc/network/interfaces || fail
fi
```
While it is correct and should work, it is a rather odd location for the `wpa_supplicant.conf` file

Mausy5043, 2015-12-29:
If you want to start moving code around, think twice. No, think three times. You might be opening a boar's pit :grinning: .  
You're probably aware that there are internal dependencies. This could take a lot of effort.

Just saying.

diederikdehaas, 2015-12-29:
Location of the file. `/boot/config/wpa_supplicant.conf` isn't the 'normal' location, that would be `/etc/wpa_supplicant/wpa_supplicant.conf`

diederikdehaas, 2015-12-30:
A general thought (of mine):
_If you're afraid to change code because it might brake something, your code is already broken. 
It just hasn't fallen apart **yet**_

goranche, 2016-03-13:
it would be better if we added a `/config/interfaces.d` directory, and copied over all files found in that location to `/rootfs/etc/network/interfaces.d`
and yes, I agree that wpa_supplicant.conf should **never** be copies over `/etc/network/interfaces` as the two are different beasts...

it's not so much the location in `/config` that worries me, but where it get's copied to (overwriting the interfaces file)

Mausy5043, 2016-03-22:
How about : 
```
/bootfs
+ config
  + apt
  + installer
  - overlay
    - etc
      - wpa_supplicant.conf
```
Copy the contents of `/bootfs/config/overlay` to `/rootfs`. (I think this was discussed before, although not my method of choice).
`wpa_supplicant.conf` tends to be installed in `/etc` together with `/etc/wpa_supplicant`.

diederikdehaas, 2016-03-22:
> copied over all files found in that location to /rootfs/etc/network/interfaces.d

Never thought about putting it there as it's also a non-standard location, but it's certainly not unlogical.

> wpa_supplicant.conf should never be copies over /etc/network/interfaces

Correct, but `/etc/network/interfaces` should contain a reference/link to the wpa_supplicant.conf, which is what the quoted code does :wink: 

>  wpa_supplicant.conf tends to be installed in /etc together with /etc/wpa_supplicant.

It may have been in the past, but it's not the default location anymore.
In `/usr/share/doc/wpasupplicant/README.Debian.gz` you can see various examples where the path to wpa_supplicant is `/etc/wpa_supplicant/wpa_supplicant.conf` and that's also the location where I'd expect to find it.

Mausy5043, 2016-06-20:
I am led to believe that this issue was solved by the merging of PR #397 into `v1.1.x`. ;-)
As a result of that belief, I am now closing this issue.
If you feel the closure was in error, please, feel free to re-open the issue.

closed 2016-06-20 by Mausy5043