debian-pi/raspbian-ua-netinst #354: Clarification | --flavour=minimal and ${minimal_packages}

aristosv, 2016-01-05:
I have been going through the rcS file, trying to better understand how it works, and reached the point where the [packages code starts](https://github.com/debian-pi/raspbian-ua-netinst/blob/master/scripts/etc/init.d/rcS#L312)

I noticed that even though we have the three presets for packages (base, minimal, server), there's also a --flavour option which is included no matter which packages preset you select. Which is a bit confusing, because --flavour=minimal has the same name as the minimal preset, and the option --flavour=minimal includes ${minimal_packages}.

So I guess my question is, what is the difference between --flavour=minimal and ${minimal_packages} ?
And what other options can I use for --flavour= ?

Thanks in advance.

Mausy5043, 2016-01-05:
Try [this link](http://bfy.tw/3ZUV). :stuck_out_tongue_winking_eye:

aristosv, 2016-01-05:
thanks. I though it was something related to this script, not debian.

closed 2016-01-05 by aristosv

Mausy5043, 2016-01-05:
no problem. :+1: