debian-pi/raspbian-ua-netinst #355: disabling services in post-install.txt

aristosv, 2016-01-09:
What is the proper way of using "systemctl" in post-install.txt? I want to disable a couple of services, and used the code below in post-install.txt, but the services were not disabled.
```
echo -n "Disabling Services..."
  chroot /rootfs /bin/systemctl disable systemd-random-seed
  chroot /rootfs /bin/systemctl disable systemd-readahead-collect
echo "OK"
```

diederikdehaas, 2016-01-09:
Do those commands work on the installed system? I'm guessing the answer is NO.
`systemctl list-units` lists the first one as `systemd-random-seed.service` ... and I couldn't find the second one.

The syntax you used is correct to execute a command in `post-install.txt`. 
As this is not a systemd support project, I'm now directly closing the issue directly.

aristosv, 2016-01-09:
These commands don't return an error when run on an installed system, but do return an error in the installation log file when run from post-install.txt. fyi.

diederikdehaas, 2016-01-09:
it could be that that command only works on a (normal) running system

kpfleming, 2016-01-11:
```systemctl``` talks to the running copy of systemd directly, it does not modify the filesystem itself. Since there is no copy of systemd running during the installation (and even if there was it wouldn't be using the same set of files as the installed system), this is unlikely to work during the installation. It will have to be done during the first boot of the system.

diederikdehaas, 2016-01-11:
Thanks for the explanation @kpfleming :+1:

Mausy5043, 2016-01-11:
Reopening this issue, as I think we need to discuss this a bit further.  

I'm concluding from the above that at some point a standard mechanism to do all the "at-first-boot" stuff will have to be implemented in the installer... People will want to enable/disable services or install extra packages, unless we can figure out how to do that "pre-first-boot".

In this specific case "simply" uninstalling the owner of the service is not an option since `systemd` is the owner:
```
pi@rbian:~ $ dpkg -S /lib/systemd/system/systemd-random-seed.service
systemd: /lib/systemd/system/systemd-random-seed.service
pi@rbian:~ $
```

How about deleting the `.service` file? (Just a thought)

Mausy5043, 2016-01-11:
@diederikdehaas : FYI both services are there on a stock system:
```
pi@rbian:~ $ systemctl status systemd-readahead-collect
● systemd-readahead-collect.service - Collect Read-Ahead Data
   Loaded: loaded (/lib/systemd/system/systemd-readahead-collect.service; disabled)
   Active: inactive (dead)
     Docs: man:systemd-readahead-replay.service(8)

pi@rbian:~ $ systemctl status systemd-random-seed
● systemd-random-seed.service - Load/Save Random Seed
   Loaded: loaded (/lib/systemd/system/systemd-random-seed.service; static)
   Active: active (exited) since Sat 2016-01-09 09:01:32 CET; 2 days ago
     Docs: man:systemd-random-seed.service(8)
           man:random(4)
  Process: 148 ExecStart=/lib/systemd/systemd-random-seed load (code=exited, status=0/SUCCESS)
 Main PID: 148 (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/systemd-random-seed.service

Jan 09 09:01:32 rbian systemd[1]: Starting Load/Save Random Seed...
```

diederikdehaas, 2016-01-11:
> both services are there on a stock system

Ah ok, I checked it on my Debian Sid machine and didn't have it there.

> People will want to enable/disable services or install extra packages, unless we can figure out how to do that "pre-first-boot".

I'm not convinced _we_ should provide/implement that.
If you want to install packages/services, you can do so via the `packages` parameter. If you want to disable a service, you shouldn't install the service in the first place. And ppl can further customize it through `post-install.txt`.

As for the 2 specifically mentioned services, I think it would be VERY unwise to disable it. 
See [here](http://www.freedesktop.org/software/systemd/man/systemd-random-seed.service.html), [here](http://www.unix.com/man-page/all/8/systemd-readahead-collect.service/) and [here](https://raspberrypi.stackexchange.com/questions/2053/would-systemd-readahead-be-beneficial-to-a-sd-card) as to why it's very beneficial to systems in general and the Pi particularly.

And then there's the thing that when we start to provide support for post-install things, where does it end? And if you do that you should support wheezy, jessie (and stretch) and also with sysv-init, systemd and upstart and also in every combination. Sounds like a dangerous path to go down to.

Of course I/we would consider Pull Requests implementing that :wink:

aristosv, 2016-01-11:
These services don't work on a read-only file system anyway. That's why I wanted to disable them. To stop them from trying to write. As for supporting post-install things, its totally up to you of course.

Mausy5043, 2016-01-11:
OK. Understood and agree. I'll leave it at that. Re-closing this issue. Please re-re-open if the need arises :grinning:

closed 2016-01-11 by Mausy5043