debian-pi/raspbian-ua-netinst #356: Feature request: Arch support

rottenbub, 2016-01-20:
Love this project, has made rapid development much easier.

But sadly I'm running into alot of walls with raspbian only containing much older version of a collection of software. So I'm starting to look at Archlinux, but really like this system as an installation process. I'm not quite savy enough to figure out how to apply this to Arch, and I'm sure some of the scripting would not match up. I know it sounds like a bunch of work, but it would definitely make my day to be able to use more up-to-date packages rather than having to figure out how to compile my own.

diederikdehaas, 2016-01-20:
The installer doesn't yet support directly installing Raspbian [Stretch, which is currently 'testing'](https://wiki.debian.org/DebianReleases#Current_Releases.2FRepositories), the raspbian.org repository does support it*. So what you could do after the initial installm is changing `jessie` to `stretch` (or `testing`) and do `apt-get dist-upgrade`.
When a new package/version enters Debian Stretch/testing, it's usually available in Raspbian Stretch/testing rather quickly. (It's mostly an automated process)

That is likely the best I can offer. 

I have no experience with Arch, apart from their wiki which simply rocks :+1: , and the chances of that changing any time soon is rather slim. That's apart from assessing and implementing the changes to the installer that need to be made in order to support that, which I'd guess would be substantial. 
And then there's also the question whether it would be wise to try to do this (feature creep).

*) The raspberrypi.org repo doesn't support Stretch, but they don't really follow Debian's/Raspbian's _Stable_ policy, so it's probably not that much of an issue. And by far most packages are provided by the raspbian.org repos.

JedMeister, 2016-01-22:
As Arch uses a different package management system I imagine that Arch support would require significant changes. Essentially a different installer...

Out of interest, do you actually need newer packages? Or are you a long term Windows user who has been trained to think that you need newer packages? :stuck_out_tongue_winking_eye: 

Obviously if there are new features you want or need that's fair enough. But there is rarely a need to update for security reasons. It seems that there are [some limitations in Raspian](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=54442), but they still (mostly) backport the security patches produced by Debian. The Debian packages are generally really well tested so you can safely enable auto security updates (and I would assume that this applies to Raspbian too). Even though you may have an old version, (not withstanding the above noted limitations) it should be just as secure as a newer version. In my experience using Debian stable reduces maintenance and provides rock solid stability. And with auto updates enabled it's not uncommon for Debian to release a security update before a vulnerability is even public!

If you are using it for playing, or a specific purpose that requires a newer version that's totally reasonable and there are many ways you can work around that if you are willing. E.g. as mentioned; upgrading to Stretch is certainly one. Or just updating specific packages from Stretch by pinning them is another. Also often upstream software producers have deb packages and/or repos that you can add. Finally; while at first glance it can seem a little scary installing from source is often pretty straight forward (although YMMV depending on the software).

IMO if you are using your Pi as a workhorse server then the stability and reduced maintenance that you get from using "stable" is well worth having to make do with older versions. Personally I still run Wheezy on my workhorse laptop and I do have a few packages that I install from upstream; mostly it all just works - and never misses a beat! :smile:

rottenbub, 2016-01-27:
Ahhh needing functionality from newer releases of nodejs(theres a solution for this one actually), mongodb(majorly), redis(somewhat), elasticsearch(notsomuch) and ansible(somewhat).

Trying to build my own version of a project someone at school is doing, he said his biggest obstacles have been needing to write code to compensate for the older versions of those packages that he's not used to.

The major part of his project so far is that it is a self-installing and self-healing cluster, to run his smart-home software(which he recently started on in nodejs).

In his version, he just flashes the latest build of raspbian-ua-netinst to a new rpi, connects it to the network and turns it on. One of the existing rpi's is periodically scanning the network for new(unknown) rpi's, ssh's to it with the default password, give it a new installer-config and post-install file, and triggers a reboot. After the rpi reinstalls again, it is automatically enrolled in the cluster, the next stage of software is installed and services are automatically rebalanced depending on detected resources(rpi1 vs rpi2, size and type of usb storage, usb printers/wifi/bluetooth/keypads, connected arduinos, hdmi displays). If an rpi is gone for too long, the services get rebalanced again and various features(printers/etc) will be disabled from the web gui. Each node has a 1-click reinstall, may take an hour or two but it always works.

So far I've got the network scanning functionality working from a raspbian perspective, and I just installed an archlinux rpi and was able to duplicate that behavior there too. From reviewing the install steps of this project and the install instructions from arch, I actually came up with a theoretical way(working too much) to convert an installed raspbian rpi to an identical clone of my arch rpi(which contains a startup script that changes the hostname to be unique).
My two issues with doing this setup is I haven't figured out yet how to convert a root install to a usb device like raspbian, and I don't get 1-click reinstall.

I do actually prefer debian environments, but know that I'm going need a newer version of nodejs and mongodb... So a bit stuck with figuring out the arch setup.... or figuring out jenkins to build newer versions of nodejs and mongodb for arm and deploying them to my own apt server...

goranche, 2016-01-27:
> In his version, he just flashes the latest build of raspbian-ua-netinst to a new rpi, connects it to the network and turns it on. One of the existing rpi's is periodically scanning the network for [...]

you might be better off creating a custom image with a ```online_config``` option in the installer-config.txt file, that way you can do with a single install pass and read the post-install script from the network... will be quicker and easier on the network

also, "need newer versions of ..." is a bad reason to switch distributions, IMHO (in this case, and yes, I'm aware I don't know the specifics)... from what you've listed, only nodejs seems like a package that would have significant changes between versions that are available under arch debian, and nodejs can be installed like this (or run as root without sudo, if that's the way you roll):
```
wget http://node-arm.herokuapp.com/node_latest_armhf.deb
sudo dpkg -i node_latest_armhf.deb
```
(if you need gpio ports, there are additional steps, but nothing too serious)

all this can be done in post-install.txt (or the network variant), which I actually do quite regularly...

also, given your application, I'd look at other options altogether... Chef, Puppet, ... (yes, you mentioned ansible, but I have a feeling you're not using it to it's full extent, given that you do multiple install passes)

> usb printers/wifi/bluetooth/keypads, connected arduinos, hdmi displays

uhm... all this in a cluster? what kind of cluster is this? :smile:

diederikdehaas, 2016-01-27:
> or figuring out jenkins to build newer versions of nodejs and mongodb for arm and deploying them to my own apt server

Which is essentially [backport](http://backports.debian.org/)-ing newer releases to stable and is a proper and often used technique to get newer software on a stable distribution.

Mausy5043, 2016-02-07:
This is the **raspbian**-ua-installer repo. Raspbian is based on Debian. Arch is not. The differences between Arch and Debian are many.
Therefore it is very unlikely that support for Arch is ever going to be included in this repo.
An upgrade path has been suggested towards `stretch`, that might work without the need to move to Arch. `stretch` is however still in testing and thus may be unstable; YMMV.
Closing this issue as the requested feature will not likely be included.

If you believe I closed this issue in error, please re-open.

closed 2016-02-07 by Mausy5043

diederikdehaas, 2016-02-07:
> Closing this issue as the requested feature will not likely be included.

Good call :+1: 
I was just thinking along similar lines wrt u-boot support. Will close that one soon (including description/explanation).