debian-pi/raspbian-ua-netinst #357: documentation should be updated

oupala, 2016-01-30:
The paragraph [Reinstalling or replacing an existing system](https://github.com/debian-pi/raspbian-ua-netinst/#reinstalling-or-replacing-an-existing-system) of the main README file seems not to be up to date.

I can't find `kernel_install.img` in my boot partition. I can only find `kernel-rpi1_install.img`, `kernel_install.img` and `kernel_install.img`.

Same problem for `installer.cpio.gz` which is not present in my boot partition. I can only find `installer-rpi1.cpio.gz` and `installer-rpi2.cpio.gz`.

The README file should be adapted to say that the filename should be adapted depending on which raspberry you are using.

diederikdehaas, 2016-01-30:
You're absolutely right, thanks for bringing it to our attention :+1:

diederikdehaas, 2016-02-21:
Fixed with the merge of https://github.com/debian-pi/raspbian-ua-netinst/pull/369

closed 2016-02-21 by diederikdehaas