debian-pi/raspbian-ua-netinst #358: Raspbian Kernel  4.1.13+  Rasbian-ua-netinst Kernel  3.18.0-trunk-rpi

4920441, 2016-02-10:
Hi,

I just updated a older install of rasbian (not installed by the ua-netinstaller)  and was really happy I got flawlessly 

`Linux lanlesspi 4.1.13+ #826 PREEMPT Fri Nov 13 20:13:22 GMT 2015 armv6l GNU/Linux`

installed as default after upgradeing...

On another raspberry pi 1  I wrote the ua-netinst image  on a sd card, hoping I would be more up to date due to the netinstall feature... but wait:

`Linux lonelesslanpi 3.18.0-trunk-rpi #1 PREEMPT Debian 3.18.5-1~exp1+rpi19 (2015-08-08) armv6l GNU/LinuxLinux 
`

Okay... for whatever reason it's stuck on 3.18.0 - so upgrade should be easy - but it isn't... despite I've got absolutely same repositories, the max I can get on the netinstalled PI is the 3.18 Edition...

Any workarounds? Is the Kernel not pulled out of the repositories?

Thanks a lot!

Cheers,

4920441

diederikdehaas, 2016-02-10:
Install the `raspberrypi-bootloader` package to get the 4.1.13+ kernel and remove/disable the `kernel` and `initramfs` lines in `/boot/config.txt`

Mausy5043, 2016-02-10:
We really should put that in the docs somewhere ;-)

goranche, 2016-02-10:
maybe start a new doc, FAQ :wink:

diederikdehaas, 2016-02-13:
I think we can conclude that this issue is handled.

closed 2016-02-13 by diederikdehaas

herrernst, 2016-07-20:
This should be put in the README. Had to search this issue to find how to update the kernel.