debian-pi/raspbian-ua-netinst #373: Security - External Files / SSH Keys

bo2k3, 2016-02-22:
Hi,

I'm a big fan of the raspbian-ua-netinstaller and I'm not sure if this is the right way to post this.
If not please excuse!

I do have a webserver that generates a custom install-config.txt and puts it into an iso which then is used to prep the SD Card. That works all like a charm.

For security reasons I wanted to accomplish two things:
1) Once the system has finished installing it should feed back to my webserver. Ideally I would have liked to issue a mysql command back somewhere along the headless installation process. So far I try various things with the post-install.txt but it tells me I cannot run/find the mysql command (I guess that's a limitation of busybox).

2) When creating the custom config on the webserver I'm creating SSH Keys and i wanted to place them along into the install image and pass them into the roots home directory. (As well some other SSH-Keys for a secure SSH to SSH connection). How could this be done?

Has anyone experiences on how to manage that?
Again, if that's an inappropriate way to ask this please disregard this topic.

Many thanks for any pointers!

goranche, 2016-02-22:
> I'm not sure if this is the right way to post this

given that there aren't better tools on github for asking questions (currently, at least), it'll have to be the right way  :wink: 

regarding the mysql command, of course, it's not present in the installer... and probably never will be, it just doesn't make sense to include it... on the installed system, though, you can install it, of course... you could also run mysql from the (partially) installed image... for example, something like this (in post-install.txt):
```
chroot /rootfs apt-get -y install mysql
chroot /rootfs /usr/bin/mysql -u username -p password server < commands.sql
```
(I'm writing this from memory, check paths, packages names, ...

regarding 2... you can access files on the SD card in the /bootfs directory... so if you copy your key to the SD card, you can just copy it to the installed system (make sure to adjust permissions and stuff, create needed directories, users, ...) like this (again, from memory, in post-install.txt):
```
cp /bootfs/id_rsa.pub /rootfs/users/username/.ssh/
```

I hope this helps...

bo2k3, 2016-02-22:
Hi,

I'm not too familiar with GitHub as you can probably understand now :)

I'll checkout your solutions with the post-install method you mentioned.

Just out of curiosity would it be possible on #2 to issue the command in /etc/init.d/rcS in the installer.cpio?

Cheers

goranche, 2016-02-22:
> would it be possible on #2 to issue the command in /etc/init.d/rcS

In theory, you could build a custom cpio with a custom rcS script in it, but I'd advice against that...

just put any command you'd like to execute in `post-install.txt` (this file get's "executed" like a script during the install process by rcS, so it's almost the same thing)

goranche, 2016-02-22:
you can take a look at this (relatively simple) [post-install.txt](https://github.com/goranche/raspbian-ua-netinst-ro/blob/master/post-install.txt)... it performs both operations, installing a debian package (unionfs-fuse) and copying over a file (a script named mount_unionfs)

bo2k3, 2016-02-22:
Alright! Messing with the cpio is not my plan anyhow :)
So doing the tricks over the post-install.txt seems to be perfect. I'll give it a try tomorrow morning.

Again thanks for your immediate inputs here!!! Really appreciated!
Besides: If you would have any ideas how to make an installation more trustworthy I'm all yours :)

diederikdehaas, 2016-03-05:
> Besides: If you would have any ideas how to make an installation more trustworthy like encrypting the contents of the SD card or so I'm all yours :)

There is already an issue open for that: https://github.com/debian-pi/raspbian-ua-netinst/issues/200
But it's not easy to implement ... and that's probably an understatement :wink:

Mausy5043, 2016-03-20:
Assuming the issue is resolved I'm closing.
If you feel the issue was closed too soon, please feel free to re-open.

closed 2016-03-20 by Mausy5043