debian-pi/raspbian-ua-netinst #374: Help with post-install.txt

djerome, 2016-02-25:
I am trying to overwrite /etc/resolv.conf so that name resolution works properly in my network.  The installer-config.txt generates this resolv.conf:

nameserver 192.168.128.35

But I need:

domain example.com
nameserver 192.168.128.35
nameserver 192.168.128.36

To get around this, I have created a file called resolvconf that is copied on to the SD card and then I try to copy this file to /etc/resolv.conf in post-install.txt e.g.:

cp /bootfs/resolvconf /etc/resolv.conf

However, when I log in to the Pi, I see the file has not been changed.  Am I doing this right or is there a better way?  How can I change /etc/resolv.conf?

Also, I am trying to use wget but can't seem to find the file.  The commands I'd like to execute are:

cd /tmp/
wget http://apt.puppetlabs.com/puppetlabs-release-jessie.deb
chroot /rootfs dpkg -i /tmp/puppetlabs-release-jessie.deb

I seem to be able to get the file but I receive the following error in the log:
Connecting to apt.puppetlabs.com (192.155.89.90:80)
^Mpuppetlabs-release-j 100% |*******************************|  7368   0:00:00 ETA
dpkg: error processing archive /tmp/puppetlabs-release-jessie.deb (--install):
 cannot access archive: No such file or directory
Errors were encountered while processing:
 /tmp/puppetlabs-release-jessie.deb

What am I doing wrong?  Where did the file end up?

diederikdehaas, 2016-02-25:
> To get around this, I have created a file called resolvconf that is copied on to the SD card and then I try to copy this file to /etc/resolv.conf in post-install.txt e.g.:
>
> cp /bootfs/resolvconf [...]

What you have done here is written that file into the filesystem of the the installer itself, but that is gone once the installation is finished. During the installation there is a directory available that represents the filesystem that will be used in the installed system and that is `/rootfs/`.
So if you change your command to the following it should work:
```
cp /bootfs/resolvconf /rootfs/etc/resolv.conf
```
Note that when using DHCP, that file is overwritten by default when the client/Pi requests an IP address from the DHCP server.

>Also, I am trying to use wget but can't seem to find the file. The commands I'd like to execute are:
>
>cd /tmp/
>wget http://apt.puppetlabs.com/puppetlabs-release-jessie.deb
>chroot /rootfs dpkg -i /tmp/puppetlabs-release-jessie.deb
>
>What am I doing wrong? Where did the file end up?

When you `chroot` into a directory, that directory becomes the new root and any file outside that directory is inaccessible. So you should copy your deb file in a subdirectory of `/rootfs/`.
I also (strongly) suggest that you use full paths when invoking commands as the $PATH you're used to having, may not be present. The following should work:
```
cd /rootfs/root/
wget http://apt.puppetlabs.com/puppetlabs-release-jessie.deb
chroot /rootfs /usr/bin/dpkg -i /root/puppetlabs-release-jessie.deb
```
HTH

djerome, 2016-02-27:
Thank you.  That worked like a charm and allowed me to do some other customizations.  Thanks for the great installer.  Combined with puppet, I can perform a nearly completely automated configuration of my Raspberry Pi.  My only regret is that some of this information wasn't in the documentation.

closed 2016-02-27 by djerome

Mausy5043, 2016-02-27:
@djerome : Glad we could help.

> My only regret is that some of this information wasn't in the documentation.

This repository is provided totally by volunteers with limited time available. If you feel the documentation is lacking or any other part could be improved we appreciate it very much if you would like to help out by providing us with your proposed changes/additions in a PR.

See the [CONTRIBUTING.md document](https://github.com/debian-pi/raspbian-ua-netinst/blob/master/CONTRIBUTING.md) for details on how to help us.