debian-pi/raspbian-ua-netinst #375: Question - RaspberryPi 3 Compatibility

aristosv, 2016-02-29:
With the new Raspbery Pi 3 on order, I was just curious. Will this script will be compatible with the new RaspberryPi 3? Its still a 32Bit Jessie installation, so I'm assuming it will be.

goranche, 2016-02-29:
we believe it will be compatible in terms of the current functionality of the script, yes... but of course, this will have to be tested... :wink: 

regarding the additional functionality (WiFi and bluetooth support), this will have to be addded, of course, and we'll try to get this sorted as soon as possible... of course, if you're able to help, that would be appreciated :+1: 

I for one hope I'll be able to get my hands on the raspberry soon (Slovenia isn't the best country to order stuff like this to :unamused: )

aristosv, 2016-02-29:
I would be more than happy to help. I'll have 9 Pi's tomorrow. So, whatever's necessary.

goranche, 2016-02-29:
> I'll have 9 Pi's tomorrow

uuuu, I hate you... :grin:

goranche, 2016-02-29:
but on a more serious note, great, thanks... :+1:   we'll see what needs to be done, and let you know...

kpfleming, 2016-02-29:
Unless they chose a really weird WiFi device for the Pi 3, the current installer should just work since it has WiFi support. Anyone who wants to use Bluetooth during the installation is just crazy :-)

diederikdehaas, 2016-02-29:
If the drivers are part of the kernel tree (that this installer uses), then it should work OOTB.

I'll test it tomorrow when my Pi 3 comes in :-D

Mausy5043, 2016-02-29:
https://osmc.tv/2016/02/raspberry-pi-3-announced-with-osmc-support/
OSMC already supports it. The BCM2837 *is* 64-bit.  WiFi/BT is provided by a BCM43438 chip.

Zeerix, 2016-02-29:
> Anyone who wants to use Bluetooth during the installation is just crazy :-)

What about a bluetooth keyboard for troubleshooting? :-)

kpfleming, 2016-02-29:
It looks like the Raspberry Pi Foundation hasn't even published drivers for the Bluetooth interface yet, so that seems to be a moot point... but good lock pairing a keyboard to the Pi while the installer is running, with no other active input devices!

diederikdehaas, 2016-02-29:
> OSMC already supports it

I've tried looking up what changes are needed to support the BCM43438 chip, but I've had no luck thus far.

ericterpstra, 2016-03-11:
I just tried in a rPi 3 and got a rainbow screen.  Tried two different power adapters and two different ethernet cables. Same result.  Then tried the SD card in a rPi B+ and it worked fine.

goranche, 2016-03-11:
> got a rainbow screen

yes, I've seen that as well (and @diederikdehaas before me), but haven't had the time to work on this yet... :blush:

Mausy5043, 2016-03-11:
The rainbow screen is displayed after/by `start.elf`. Next:
- [ ] `config.txt` is read
- [ ] `cmdline.txt` is read
- [ ] the kernel `*.img` is loaded

So, somewhere in that list something goes off.

diederikdehaas, 2016-03-11:
It goes wrong with loading of the kernel.
For starters you MUST get new firmware files as there is now a DeviceTree file for the Pi 3, but I've noticed there are also other files which are not present in the 'latest' `raspberrypi-bootloader-nokernel` package.
For users of a serial console, you probably want to take a look at https://github.com/RPi-Distro/repo/issues/22

Solear, 2016-03-26:
Mh, it still does not work :-(
Any solution?

goranche, 2016-03-26:
no, it doesn't... :unamused:   I've had some luck building a completely custom installer (by hand), using the `next` from from the [raspberry firmware repository](https://github.com/raspberrypi/firmware), and was convinced they'd moved it to the `master` branch (feeling hopefull it would be released, so we could use it), but going through the same (manual) process of building a custom installer from the master branch resulted in an installer that can't mount the filesystem...

well, at least the kernel boots... 😔

so, we're working on it, but at this time, no real solution has been found yet... I'll at least try to put together a custom image, so there'll be a workaround... (might even happen today, not sure)

mattotodd, 2016-04-07:
Any update here? 

@goranche - by "some luck" do you mean you were able to boot a pi without issues using `next`, or even with `next` there were other hurdles (i realize bluetooth would not be working out of the box) ?

goranche, 2016-04-07:
no real update, no, sorry... I'm "a bit" short on time, have a huge deadline coming up, and 2 of my team members sick, so it's ... let's stick with fun :unamused: :innocent: 

I was able to boot the RPi3, and get the installer to run, but then had to jump through a lot of hoops to get the installed system to run (basically copy over the `next` firmware into `/boot`)... but it was a gamble, didn't get it running reliably, and then (as mentioned) ran out of time...

there was some "chatter" about a new kernel (and firmware), but I'm not sure where that's at...

bwkihxev, 2016-04-11:
I've developed a base image from a raspbian-ua-netinst install on a RPi2.    

Let's say I want to start using RPi3 instead, is there a package that can be updated for the RPi2 image to boot on the RPi3? ATM I am getting a rainbow screen simply moving the MicroSD from Pi2 to Pi3

diederikdehaas, 2016-04-11:
@bwkihxev install the `raspberrypi-bootloader` package as that contains the latest kernel + firmware.
This will remove some other packages, but that's ok.

4eMaLo, 2016-05-10:
I got it to work for me by copying the boot/ directory from [the official reposiory](https://github.com/raspberrypi/firmware/) into the SD root and adding 

`[pi3]`
`initramfs installer-rpi2.cpio.gz`

to config.txt

cherylquirion, 2016-05-26:
Just to clarify @4eMaLo instructions a wee bit.  

On my computer, I copied all of the contents of the boot directory from [the official repository](https://github.com/raspberrypi/firmware/) onto the SD root on the memory card, including start.elf, start_cd.elf, start_db.elf and start_x.elf. Just replace everything that exists on the SD with the files in the boot directory of the official release. 

Then I edited the config.txt file on the memory card and added these lines as instructed:

`[pi3]`
`initramfs installer-rpi2.cpio.gz`

Next, I put the memory card in the Pi and booted it up. It is happily installing now. Thanks for the tip @4eMaLo. I am about to order 100+ Pis and want to be able to use the Pi3.

mattotodd, 2016-05-26:
thx @cherylquirion -  i'll going to try the same later today.

Will the same image work for both pi2 and 3?  or is this pi3 specific ?

cherylquirion, 2016-05-26:
Not sure. I have to test that part yet. Will report back.

diederikdehaas, 2016-05-26:
> Will the same image work for both pi2 and 3? or is this pi3 specific ?

@mattotodd For the Pi 2 you need to use the `[pi2]` header, but I have not reason to believe it won't work on the Pi2 ... mostly because it already worked on the Pi2 :wink:

mattotodd, 2016-05-26:
@diederikdehaas  - i know it does, because i've used it 

I meant, if i were distributing an image to end users, would they need to know which pi they had, and pick the correct image,  or can i just supply a single image.

eLvErDe, 2016-05-27:
Hi guys,

RPi2 are slowly running out of stock so I guess this issue should be addressed asap. In the meanwhile, as the firwmare git repo is huge, I made a small TGZ containing the file needed only:
http://metz.le-vert.net/pub/rpi-firmware-boot-folder-20160527.tar.gz

Be carefull, there's no subfolder so it's aimed to be extracted to /boot/ directly. The installer is running fine for the moment so once finished, I plan to extract this archive again on /boot and check the config.txt. Finger crossed....

Best regards, Adam.

goranche, 2016-05-27:
it's pretty easy to get the installer running on the RPi3 in a manual fashion, and if you know it'll run on the RPi3 only... (and even then it's an ugly solution)
if, however, you want to do it "the right way", it becomes a much (**MUCH**) bigger problem...

that said, of course, time is being spenf on getting it right...

eLvErDe, 2016-05-27:
No worry, I fully agree but is that much more than updating a firmware deb package with new code and change template stuff to generate config.txt with pi3 support ?
Do you also need to update the kernel package ?

Adam.

Btw, regarding config.txt AFTER installation I guess it makes more sense to use "initrd.img-3.18.0-trunk-rpi2" instead of the installer one.

goranche, 2016-05-27:
yes, the kernel package is one of the (bigger) issues...

eLvErDe, 2016-05-27:
Speaking about the kernel.... More workaround if you want to use additional modules (ie: i2c):

> cd /tmp/
> wget http://metz.le-vert.net/pub/rpi-firmware-lib_modules-folder-20160527.tar.gz
> tar xvzf rpi-firmware-lib_modules-folder-20160527.tar.gz -C /lib/modules/
> depmod

Sadly Pi foundation provided kernel does not include the watchdog module, but at least i2c seems to work fine.

diederikdehaas, 2016-05-27:
> Sadly Pi foundation provided kernel does not include the watchdog module,

It's not a module anymore, but it is build in :disappointed: 
Modprobe-ing it actually seems to cause issues (but I don't know the details of that).

Mausy5043, 2016-05-27:
Same goes for the HWRNG module `bcm2708-rng`. Which is also built-in for current (4.something) kernels.

diederikdehaas, 2016-05-27:
> Same goes for the HWRNG module bcm2708-rng.

Oh crap! I actually meant that one :innocent: 
Wouldn't be surprised if the watchdog module is now also build in, but haven't checked (yet)

willemdh, 2016-05-31:
I followed @4eMaLo and @cherylquirion 's procedure. The Rpi3 installed fine, but I did notice one thing. The total physical memory is too low.

On rpi2:
free
```
             total       used       free     shared    buffers     cached
Mem:        949488      74692     874796       4424       7516      33224
```
```
cat /proc/meminfo
MemTotal:         949488 kB
MemFree:          874228 kB
MemAvailable:     904504 kB
```
On rpi3:
```
free
             total       used       free     shared    buffers     cached
Mem:        120208     118184       2024       3920      18120      67124
```
```
cat /proc/meminfo
MemTotal:         120208 kB
MemFree:            1964 kB
MemAvailable:      88180 kB
```

As you can see tot mem on the rpi3 is far too low. So is it me alone? Or does someone else sees this too?

4eMaLo, 2016-06-02:
@willemdh 
I've checked `/proc/meminfo` on my RPi 3 and it sais I have 947748 kB memory in total. Did you check `gpu_mem` in `/boot/config.txt`? if this is ok too, try burning the official image to the SD card and then check the memory again. If it is still too low your Pi might actually be broken..

Edubits, 2016-06-22:
@willemdh and @4eMaLo, I had exactly the same with my first try! I followed the steps, downloaded the boot files from the official repo (version 1.20160620) and tried to install. It did install, but afterwards didn't come past the rainbow screen (also not after replace the start* files). I could login through SSH, and saw the same amount of memory. 
Then I tried again, but now using the tar by @eLvErDe instead of the download from the official repo. That worked as expected, I even didn't have to replace the start* after a reboot. So I assume whatever difference there is between these boot files is causing the wrong amount of memory.

willemdh, 2016-06-22:
@Edubits Thanks for letting us know. I'l also try again with the tar by @eLvErDe

CheshireCat44, 2016-06-24:
Firstly, thanks to the core team for creating this very useful software. Secondly, thanks to @cherylquirion and @eLvErDe for enabling me to get this working on a Pi3. 
I copied the contents of the 1.0.8.1 zip to the microSD card, copied the contents of @eLvErDe's firmware.tgz on top and then added my own installer-config.txt, post-install.txt and other modifications.
My initial attempt failed with the log showing: 
Copying /boot files in... OK
Loading f2fs module... 
Oh noes, something went wrong!
but when I removed 
rootfstype = f2fs
it worked perfectly. Like @Edubits, there was no need to copy the start... files later.
With my extensions to the system, the Pi3 will perform a full unattended install of a LAMP server with Webmin and send out an email when it is done. 
Any suggestion, please, for how rootfstype = f2fs might be made to work?

alexdebiasio, 2016-07-18:
Hello,
First of all thank you for this project.
Is there any update for a version that supports out of the box the Pi3?

thijstriemstra, 2016-07-19:
+1, what is working, what isn't?

Phiber2000, 2016-07-20:
I'm using raspbian-ua-netinst since long time.

So I forked this great project based on the 1.1.x branch to get the Pi3 working in the meantime. All I had to do was replacing the rasbian kernel with the raspberrypi.org kernel and make the scripts compatible using both repositories.
I tested three different Pi models and f2fs successfully.

Other changes: raspi-copies-and-fills (for memory improvement), bash-completion and console-common are installed by default.

If someone's interested, it's [here](https://github.com/Phiber2000/raspberrypi-ua-netinst).

eLvErDe, 2016-07-20:
Hey,

Just rebased my work fork on yours and I've been able to generate a valid zip file that installed successfully on a pi2 (using pi.org kernel). So thanks a lot, it's a lot cleaner than my previous hack!

Adam.

CheshireCat44, 2016-07-20:
Many thanks to @Phiber2000 : it worked "out of the box" on a Pi3. And with f2fs 8-} 
So far I haven't tried it on older Pis.

A great project further enhanced.

diederikdehaas, 2016-08-06:
It probably won't be too long before support for the Pi 3 is available in the v1.1.x branch ... :wink:

diederikdehaas, 2016-08-09:
If one or more people could test out PR #418 that would be great :smile:

Mausy5043, 2016-08-22:
Closing this issue as I believe PR #418 solves this. If you feel this is not the case please feel free to re-open the issue.

closed 2016-08-22 by Mausy5043