debian-pi/raspbian-ua-netinst #377: online_config.txt for online install via raw githubusercontent not working.

ronnyvdbr, 2016-03-03:
Hi, i'm trying to install with your net installer, beautiful work by the way.
I do however have one issue, i'm trying to use your online installer feature, and it doesn't work.
In my installer-config.txt I have the option:
online_config=https://raw.githubusercontent.com/ronnyvdbr/RaspberryIPCamera/master/NetInstall/online_config.txt

However, on starting my raspberry pi with the net installer, I get the message, oh noes, something went wrong.

When I drop myself into a shell and execute wget https://raw.githubusercontent.com/ronnyvdbr/RaspberryIPCamera/master/NetInstall/online_config.txt
I get the message from wget that the url is not http nor ftp.
Apparently when I link from raw.githubusercontent.com I automatically get redirected to https instead of http, so I suspect this to be the problem.
Is there any work around for this?

diederikdehaas, 2016-03-03:
Thanks for bringing this up to our attention :+1: 

I had actually noticed this quite a while ago, but forgotten all about it :disappointed: 
AFAIK there currently isn't a way around it.
The 'best' alternative I can suggest for now is downloading the repository as a zip file, put that alongside the installer files and work from that.

goranche, 2016-03-03:
just a quick note... have you tried this config file with the installer? I don't think it will do what you expect it to...

the `chroot /rootfs` in the first line will not cause the whole script to be executed in the installed environment (AFAIK)

diederikdehaas, 2016-03-03:
correct, see #378 :wink:

goranche, 2016-03-03:
oh well... :blush:

ronnyvdbr, 2016-03-04:
> I had actually noticed this quite a while ago, but forgotten all about it [image:

Hi, thanks. I think I will just include the post installer file with my net
image, this should work too but is more work if changes need to be made to
this file afterwards.
Anyhow,  hope u see the opportunity to fix it in the future.

goranche, 2016-03-09:
I'll reopen this open for now, as there will probably be some work done on this soon :wink: 

@ronnyvdbr

ronnyvdbr, 2016-03-09:
oops, my appology!

hidden-refuge, 2016-05-20:
The problem here is that wget cannot identify the TLS certificate issuer because the trusted CA certificate database is missing. To make it work you have to install the package "ca-certificates" or add the option "--no-check-certificate" to all wget operations on https links.

So I see these two options to fix it:
1. Add ca-certificates to packages to be installed before online installation is performed
2. Disable certificate check for wget

Which one is better? Well, it would be cleaner to have ca-certificates just to get rid of that dumb errors and to avoid always having to use --no-check-certificate. Otherwise if you do not wish to load up the installation the second option to disable the certificate check is perfect.

As I know people... most are afraid of messages caused by tools just because the certificate is self-signed or cannot be validated due to missing CA databases. LOL I have been even called an idiot because I developed my scripts on a minimal OpenVZ VPS where ca-certificates was not installed so I was publishing links with "--no-check-certificate" everywhere to avoid issues (at that point I didn't know I was missing the CA database on my dev server). The guy who called me an idiot didn't even bother to check that my download server had a valid TLS certificate... He said he would never download from a site with an invalid certificate. Haha. So you see. Better not risk anything. In my opinion option one would be the cleanest solution.

diederikdehaas, 2019-12-11:
This has now been fixed in the v1.1.x branch

closed 2019-12-11 by diederikdehaas