debian-pi/raspbian-ua-netinst #378: customization with post-install.txt does not work.

ronnyvdbr, 2016-03-03:
Hi, when my post-install.txt kicks off it can't find any commands, like for example apt-get.
There is a folder rootfs to be found in the root.
I can chroot into it, then commands like apt-get are to be found.
I tried adding this chroot /rootfs to post-install.txt and reran the install.
But still no joy when the script kicks off.
Am I doing it wrong?

richard-scott, 2016-03-03:
From what I can see, the commands in post-install.txt are executed "outside" the chroot.
Perhaps something like this in your post-install.txt will help:

```
cat << 'EOF' > /rootfs/post-install.sh
#!/bin/bash
apt-get install axel
EOF
chmod +x  /rootfs/post-install.sh
chroot /rootfs /rootfs/post-install.sh
rm -f rootfs/post-install.sh
```

That should install the axel packege if all goes well.

diederikdehaas, 2016-03-03:
Doing something like @richard-scott suggested seems like the best way to go, but there is one minor (but important) error in it:
`chroot /rootfs /rootfs/post-install.sh` should be `chroot /rootfs /post-install.sh`

Normally, **every time** you want to perform a command on the target system, you have to place `chroot /rootfs` in front of the command.

There is one other _very_ important aspect why it likely still won't work and that is $PATH.
When you're logged in to a *nix system, open a terminal and do `echo $PATH` you will see a list of directories. One of the directories you see there is `/usr/bin/` and that's the reason that `apt-get` (for example) works when you're logged in. Because of $PATH the system 'translates' that to `/usr/bin/apt-get`.
You can fix your script by either using full paths (for everything) or start your script with setting `PATH` and then 'exporting' it with `export PATH`.

HTH

goranche, 2016-03-03:
> using full paths

this isn't needed... `${PATH}` is a list of paths to search for commands that are executed (the don't "translate"), and this list is the same for both the installer environment as well as the installed system...

richard-scott, 2016-03-03:
Thanks @diederikdehaas I've updated my example :-)
and as @goranche said, the paths in $PATH are inherited from the running system so unless you need something in a non-standard palce it should pretty much work.

ronnyvdbr, 2016-03-04:
Hi, thanks for your reply,  I was able to sort the issue by dumping my
customisations to the shell file and executing that in the chroot as
suggested in your answers.
I was under the assumption that, once the chroot command is given, all
subsequent commands would run in the chrooted environment.  This isn't true
if the parent isn't running in the chrooted environment however.

closed 2016-03-07 by ronnyvdbr