debian-pi/raspbian-ua-netinst #383: duplicate localhost entry in /etc/hosts

richard-scott, 2016-03-07:
I've noticed that in the latest 1.1.x release, that localhost is duplicated in /etc/hosts:
```
root@pi:~# cat /etc/hosts
127.0.0.1 localhost
127.0.0.1       localhost
127.0.1.1       pi
root@pi:~#
```
I can only guess that when [this](https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.1.x/scripts/etc/init.d/rcS#L769-L777) runs that there is already an entry for localhost in /etc/hosts.

Maybe line 777 needs altering to overwrite /etc/hosts instead of appending to it.

diederikdehaas, 2016-03-07:
Are you installing `wheezy` or `jessie` (if you didn't specify it, it would be `jessie`)?

richard-scott, 2016-03-07:
I ran this today:

```
git clone -b v1.1.x --single-branch https://github.com/debian-pi/raspbian-ua-netinst.git
cd raspbian-ua-netinst
./clean.sh
./update.sh
./build.sh
./buildroot.sh
dd if=raspbian-ua-netinst-20160307-gitf0a95bc.img of=/dev/sdc
```

Did not edit any files, so used defaults where applicable.

Mausy5043, 2016-03-07:
Confirmed on `jessie` Pi1B and P1B+.

Mausy5043, 2016-03-07:
I wouldn't worry too much about this. There is no negative impact on the system. It just looks less perfect.
Nevertheless, a good find. :+1:

Mausy5043, 2016-03-07:
Personally, I always add these:

```
echo "::1             localhost ip6-localhost ip6-loopback" >> /rootfs/etc/hosts
echo "fe00::0         ip6-localnet" >> /rootfs/etc/hosts
echo "ff00::0         ip6-mcastprefix" >> /rootfs/etc/hosts
echo "ff02::1         ip6-allnodes" >> /rootfs/etc/hosts
echo "ff02::2         ip6-allrouters" >> /rootfs/etc/hosts
```

diederikdehaas, 2016-03-07:
Hmm ... I really wonder how this came out to be, just like the double `lo` definition in `/etc/network/interfaces` on `wheezy`.

Mausy5043, 2016-03-07:
I'm guessing someone changed something upstream.

goranche, 2016-03-07:
I've added a check for an existing `localhost` in `/rootfs/etc/hosts` to #381

richard-scott, 2016-03-08:
Thank you very much for fixing. I only noticed this as I am using Ansible to configure my Pi and it wasn't changing ~all~ the localhost entries in the file... it was then that I realised it was duplciated.

One thing, I think you are missing a space on this [line](https://github.com/goranche/raspbian-ua-netinst/commit/9815f933fb5d977fdd48cf8af5fe9f6270292d4f#diff-ef22c8fabe7f21fa3c3ab45495d6774bR534):
```
if ! grep -q "localhost" /rootfs/etc/hosts; then
```
Shouldn't there be a space after hosts before the ; ?

kpfleming, 2016-03-08:
No, there is no whitespace required there. Welcome to the wonderful world of POSIX shell scripting :-)

goranche, 2016-03-08:
@richard-scott as @kpfleming said, whitespace isn't required at that point... and I personally like this way better... :innocent:

goranche, 2016-03-13:
This should be fixed in both [https://github.com/debian-pi/raspbian-ua-netinst/tree/v1.0.x](v1.0.x) and [https://github.com/debian-pi/raspbian-ua-netinst/tree/v1.1.x](v1.1.x) branches, but you'll have to build the installer yourself for the moment... just until a new release is created, of course ;)

Mausy5043, 2016-03-15:
PR #385 resolved this issue for v1.1.x

@goranche which PR resolved this for v1.0.x?

closed 2016-03-15 by Mausy5043