debian-pi/raspbian-ua-netinst #384: Stupid newbie question probably

ronnyvdbr, 2016-03-09:
Hi, when I boot the Raspberry foundation jessie minimal image, I have dhcpcd client package version 6.7.1.
If I use your net-installer, I get dhcpcd version 3.2.3.

I compared the apt-sources list on both, and they seem to be set to mirrorredirector.raspbian.org on both.

Can I let the net-installer install the same as the jessie minimal foundation image?

diederikdehaas, 2016-03-09:
> I have dhcpcd client package version 6.7.1. If I use your net-installer, I get dhcpcd version 3.2.3.

There are actually 2 different packages you're referring to. One is called `dhcpcd` and should only be available till `wheezy` (thus not in `jessie`). The other is `dhcpcd5` and that is what the Raspberry Pi Foundation uses and is probably the one you want.

ronnyvdbr, 2016-03-09:
D'oh!  
Thanks, I'm glad I asked, because I would be searching for days otherwise I'm afraid.
Given the similar names this is a confusing case.
Another question, why is the net-installer using a totally different kernel than the raspberry foundation image?
the foundation image gives me below specs:
Linux raspberrypi 4.1.18-v7+ #846 SMP Thu Feb 25 14:22:53 GMT 2016 armv7l GNU/Linux
while the net installer is more like version 3.18 or something
Thanks!

diederikdehaas, 2016-03-09:
> why is the net-installer using a totally different kernel than the raspberry foundation image

On a Debian system, you'll always have a `linux-image-X` and a corresponding `linux-headers-X` package. With that headers package, users can easily compile their own kernel modules, for example for a wireless network adapter which isn't supported by the kernel itself (yet). There are also various [DKMS](https://en.wikipedia.org/wiki/Dynamic_Kernel_Module_Support) enabled packages, which are really cool, but they just don't work with the RPF kernels, as they don't have the [corresponding linux-headers-X package](https://github.com/RPi-Distro/repo/issues/11).

This is probably the most important reason, but there are others, in which we have decided to stay closer to Debian then the RPF has chosen. Unfortunately, the whole of raspbian.org and all the packages that are provided via that, is run by only 1 person. And (probably) due to the lack of time, there hasn't been an update to the Debian/raspbian.org style kernel in a while.

ronnyvdbr, 2016-03-09:
Hi ok thanks, was just wondering if there was any possible upgrade path for this.
I can assume if you need to perform maintenance for such a whole repository on your own that things are a on a bit slower track.
Kudo's though for getting this all done as an individual!

closed 2016-03-09 by ronnyvdbr