debian-pi/raspbian-ua-netinst #386: Add avahi-daemon to the installed system

goranche, 2016-03-13:
Avahi daemon implements Apple's Zeroconf architecture (also known as "Rendezvous" or "Bonjour") and allows network discovery of devices running it... I'd suggest adding the daemon to the default install (or maybe preferably through a by default off option), as this makes it *much* easier to find devices once they're up and running.
Avahi would be set up to advertise it's SSH service, so discovery is sped up even more.

this in an example of it's use (on a Mac, but with different tools - avahi, obviously - it works on Linux as well, and even Windows is supported in the form of Bonjour browsers):
[image]

then it's becomes a simple `ssh <username>@pi.local` to log into the device.

goranche, 2016-03-13:
the reason I suggest using this through a by default off option is in the size of the install... avahi-daemon by itself is quite small, but the debian package lists a large number of dependencies and the it becomes a 45MB install... :unamused: 

still a useful tool, though, something I install on **all** my devices

Mausy5043, 2016-03-13:
I used to like it but I'm nolonger using avahi because I know the hostname of all devices and have no other uses for it.

goranche, 2016-03-13:
I'm running setups, where I have a lot of devices running (50+), and knowing all the hostnames is not an option... :wink: 

also, since the number of devices is that large, I use the same SD card image for all and calculate (or retrieve from a central server, depends on the setup) the hostname from the (eth0) MAC address, so it becomes a bit harder, to put it mildly :smile:

richard-scott, 2016-03-13:
As a Pi can boot without it, I don't think it should be forced on users to have this installed. Yes, it may be handy to have, but 100% of installations work fine without it. 

A better idea would be to have a post installation hook to use something like [Ansible ](http://docs.ansible.com/ansible/) to automate extra configuriaton after booting. Why [Ansible ](http://docs.ansible.com/ansible/)? Well, it requires no client agent to be installed so has zero additional footprint in a Pi apart from python-apt being installed.

goranche, 2016-03-13:
the problem with Ansible is that it's one of many solutions (avahi isn't, AFAIK), so including this would "require" doing the same for Chef, Puppet, Salt, ... to name a few :wink: 

also, having avahi around would make life easier when using Ansible, as your computer needs to connect to the newly installed devices... (this is also a major issue I have with Ansible, many of my installs have NAT connectivity, but I can't connect to them from the outside, which makes Ansible pretty much useless)

goranche, 2016-03-13:
hm... although there might be a "[http://jpmens.net/2012/07/14/ansible-pull-instead-of-push](workaround)" to Ansibles push mode of operation... :thought_balloon: 

but this would still leave the "what about the others" question... but feel free to open an issue with the request, of course, and maybe even provide a pull request with the feature implemented :+1:

Mausy5043, 2016-03-13:
Isn't the Ansible vs. Chef vs. Puppet vs. whatever  the same discussion as Debian vs. Arch vs. BSD vs. whichever would be?

richard-scott, 2016-03-13:
Fair point on installing Configuration Managemnet tools, ignore that idea as as they are not essential for installation :smiley:

As for Avahi... Did you know that you can install extra packages during installation via the "packages=" option? Maybe that is a better option than enforcing all users to have them installed.

goranche, 2016-03-13:
> you can install extra packages during installation

yes, I do know that :smile: 

but installing the package isn't enough to get avahi-daemon up and running, you usually (at least for a network with Macs in it) have to change an option in `/etc/default/avahi`, and if you want to advertise the ssh service, you have to create `/etc/avahi/services/ssh.service`...
you can do both things in post-install.txt, of course, it's the way I do things at the moment :wink:  but given how useful the service is, I though it might make sense to include it in the installer

goranche, 2016-03-13:
> Isn't the Ansible vs. Chef vs. Puppet vs. whatever the same discussion as

@Mausy5043 no, I wouldn't really agree with you on that :wink:

diederikdehaas, 2016-03-14:
I'm not in favor of this as I never use it and I'm guessing a large part of our users, wanting a minimal system, aren't either.
As I see it, it is just some random application and if we would provide options for this application, where would it end?

goranche, 2016-03-14:
> just some random application

LOL  :innocent: 

ok

closed 2016-03-14 by goranche