debian-pi/raspbian-ua-netinst #387: /etc/hostname blank after install

richard-scott, 2016-03-16:
After building the installer from  the v1.1.x branch I have an empty /etc/hostname file after rebooting., it 

Steps to reproduce:

1. git clone -b v1.1.x --single-branch https://github.com/debian-pi/raspbian-ua-netinst.git
2. build the latest image downloaded
3. dd'd a new image onto my SD card, made no other changes and booted it.

My /installer-config.txt does not exist as nothing has been changed from defaults.

richard-scott, 2016-03-16:
ok, small update. I just mounted my SD Card on my Linux box to copy files off to add to this issue and looked at /mnt/sdcard/etc/hostname, and low and behold it contains "pi"... which is odd, as when I boot the SD Card in my Pi, the /etc/hostname file is empty.

I think the dhcp client is now removing my hostname as one is not provided by my router. That didn't used to happen, and I've not changed my router.

goranche, 2016-03-16:
could you please do a `ls -l /etc/hostname` on the running system and post the result?

Mausy5043, 2016-03-16:
@richard-scott : Do you have an `installer-config.txt` that you could share? I'm also **very** interested in the logfile (probably in `/var/log/raspbian-ua-netinst.log`)
The logfile can be pushed to pastebin to avoid clutter here.

richard-scott, 2016-03-16:
My installer-config.txt is non existent as I've not set any other options.
Sure, http://pastebin.com/8UUSE7bv

Is there somewhere I can upload the .img file so you can use the exact same build I have?

Mausy5043, 2016-03-16:
@richard-scott : The logfile looks very unsuspicious :disappointed_relieved: . Could you also show us the output of:
```
ls -al /etc/hostname
```

goranche, 2016-03-16:
@richard-scott would it be possible for you to try out the changes from [this branch](https://github.com/goranche/raspbian-ua-netinst/tree/hosts-fix)? (the branch was made from the [main v1.1.x](https://github.com/debian-pi/raspbian-ua-netinst/tree/v1.1.x))

I've tried it on my device, and found that it works better (as expected), but would like to confirm, it's not just something in my environment... :thought_balloon:

richard-scott, 2016-03-17:
@Mausy5043 - My /etc/hostname file had a timestamp that was not recent, but was not the time of the install either...somewhere inbetween. I think it got created at the first reboot after install and was set as empty at that point.

@goranche  I'll test the new branch later this evening (its 9am UK time here, so won't be for 12hrs or so). From my tests last night, the /etc/hostname file was being populated with my hostname but only after I rebooted my DHCP server/router. It seems that the router may have been not giving out a hostname (as it didn't have one in its lease) and the dhcp client was using "nothing" as the entry for /etc/hostname... could that be possible?

goranche, 2016-03-17:
> its 9am UK time here

so? excuses... :grin: 
:wink:   no problem, thanks in advance for doing the test at all :+1: 

yeah, I guess the described issues might be caused by the DHCP server, although in case nothing gets returned for the hostname, it should still be using the default of the device, in this case `pi`...

anyway, I hope this fix will actually make things better... famous last words, a.k.a. "hope dies last" 🤔

richard-scott, 2016-03-17:
Is there a way to get the installer to poweroff rather than reboot?
I could then mount the SD Card on my Linux box and see the pre-dhcp edited contents of the file.

Would you be able to update your host-fix branch to get it to do that for me?

That way we will know for sure if its installer based or dhcp based as we now believe.

goranche, 2016-03-17:
you can add `final_action=halt` to post-install.txt for installers built from the `v1.1.x` branch

richard-scott, 2016-03-17:
Ta, I'd literally just noticed that in the bottom of the rcS file :wink:

diederikdehaas, 2016-03-17:
> `final_action=halt` to post-install.txt

Or in `installer-config.txt` :wink:

goranche, 2016-03-17:
oh, sh**t  🙀   sorry about that one... yes, it should be added to `installer-config.txt`, not `post-install.txt`  :blush: 

thanks, @diederikdehaas

diederikdehaas, 2016-03-17:
I haven't tried it, but I _think_ it should also work in `post-install.txt`

goranche, 2016-03-17:
I think you're right, it actually might :thought_balloon: 

but still, it belongs in the config

richard-scott, 2016-03-17:
The "hosts-fix" branch worked perfect for me:

https://github.com/goranche/raspbian-ua-netinst/commit/5065c8207b2604464879a0bbea84127fd0444ce5

Now for a full test getting the Pi to reboot at the end.

goranche, 2016-03-17:
> Now for a full test getting the Pi to reboot at the end

ok, please report the results, so we can move on from there... :wink: 

and thanks for the testing :+1:

richard-scott, 2016-03-17:
My test results are a clean boot configured as follows:

```
root@pi:~# hostname
pi
root@pi:~# hostname -f
pi
root@pi:~# cat /etc/hosts
127.0.0.1 localhost
127.0.1.1 pi
root@pi:~# cat /etc/hostname
pi
root@pi:~#
```

Looks like we can mark this one solved :smile:

goranche, 2016-03-18:
> we can mark this one solved

after we merge the branch into the main repository :wink: 

thanks for the tests (and reporting the issue, of course) :+1:

Mausy5043, 2016-03-20:
PR #388 fixes issue #387. Closing.
If you feel the issue is not fixed, please re-open.

PR #388 (2016-03-19): Fix hosts file creation
It adds a newline to `/etc/hostname` (same will be done for v1.0.x)
In addition to this, `/etc/hosts` gets generated in a different way and afterwards contains needed entries

Fixes #387
commit f94abd71 (2016-03-19): Merge pull request #388 from goranche/hosts-fix

Fix hosts file creation
Fixes #387 (v1.1.x)
closed 2016-03-20 by Mausy5043