debian-pi/raspbian-ua-netinst #395: Root or user?

Nyzo, 2016-03-27:
Hey,

Is it possible to set an user and password instead of root ?

thx!

goranche, 2016-03-27:
in the current release (and all 1.0.x version) you'll have to do this yourself in `post-install.txt`

the v1.1.x branch recognizes options `username` and `userpw`, but for the time being, you'll have to build that installer yourself... it will be released when we can stabilize RPi3 support

the method I use is this command in `post-install.txt`:
  `chroot /rootfs useradd -s /bin/bash -p <pass-hash> -m -G users[,group,...] <username>`

generating the `pass-hash` is simple enough snd shouldn't be hard to find... if you have perl available, you might try this:
  `perl -e 'print crypt("YourPasswd", "salt"),"\n"'`

goranche, 2016-03-27:
you can, of course, set the root password in the current released version... not sure if you were asking about that as well... :wink:

Nyzo, 2016-03-27:
My question was simply find out if I could not put a password for the root, and instead create a user.
I think you answered my question with this command:
chroot /rootfs useradd -s /bin/bash -p <pass-hash> -m -G users[,group,...] <username>

goranche, 2016-03-27:
in case you wish to disable the root login (it's what I do as well), I'd strongly suggest installing the sudo package and adding the created user to the sudo group:
```
chroot /rootfs apt-get -y install sudo
chroot /rootfs useradd -s /bin/bash -p <pass_hash> -m -G users,sudo <username>
```
(you could skip the first command by adding `sudo` to the `packages` option in `installer-config.txt`

Mausy5043, 2016-03-27:
> if you have perl available, you might try this:
```
perl -e 'print crypt("YourPasswd", "salt"),"\n"'
```

Please note that `crypt` only uses the first two characters of the salt.

goranche, 2016-03-27:
> only uses the first two characters of the salt

technically, yes, as [documented](http://man7.org/linux/man-pages/man3/crypt.3.html), you don't need to provide more than 2 characters. Also, the two characters are copied to the start of the resulting string (they are used to pertubate the DES algorithm). If this were not the case, you could never check the actual password...
```
> perl -e 'print crypt("YourPasswd", "salt"),"\n"'
safv8d8ESMmWk
> perl -e 'print crypt("YourPasswd", "tlas"),"\n"'
tlReguxKGkGEk
```

as an interesting thing, BSD (and with that, OS X as well) has an [extended mode](https://www.freebsd.org/cgi/man.cgi?crypt(3)), where 8 chars from the salt are used, but takes **a lot** more time to generate the salt (1.281s as opposed to 0.009s):
```
> perl -e 'print crypt("YourPasswd", "_longersalt"),"\n"'
_longersaZOmYVh7Sp0s
> perl -e 'print crypt("YourPasswd", "_tlasregnol"),"\n"'
_tlasregnihLcrKIY4gU
```

goranche, 2016-03-27:
@Nyzo are you satisfied with the answers? can we (or you) close this issue?

Mausy5043, 2016-03-27:
```
$ perl -e 'print crypt("YourPasswd", "salt"),"\n"'
safv8d8ESMmWk
$ perl -e 'print crypt("YourPasswd", "sand"),"\n"'
safv8d8ESMmWk
$
```

goranche, 2016-03-27:
yes, it's by design... a link to documentation is in my previous comment

Nyzo, 2016-03-27:
Yeah thx for help!

closed 2016-03-27 by Nyzo