debian-pi/raspbian-ua-netinst #398: Can't login to my raspberry pi after install

Viliasas, 2016-04-01:
Hello,

I'm trying to get my Raspberry Pi 1 Model B running with raspbian-ua-netinst. But after installation I try to login with root/raspbian logins and I get "Access denied" message.

As for the installation, I've formatted my 4 GB SD card with SD card formatter, and unzipped "raspbian-ua-netinst-v1.0.8.1.zip" contents to it. And after installation I can't login with root.

Maybe someone know what I’m doing wrong?

richard-scott, 2016-04-01:
How long do you wait before attempting to log in via SSH? It can take up to 30min for the first boot installation to complete... once the installer has setup your new system it will reboot and you should then be able to login.

Viliasas, 2016-04-02:
Thanks for the info. I've waited longer and now I'm able to connect.

closed 2016-04-02 by Viliasas