debian-pi/raspbian-ua-netinst #399: Execute Bash script post-install

willemdh, 2016-04-07:
Hello,

Is there some way to execute a Bash script post-install?

Grtz

mattotodd, 2016-04-07:
Yes, the [docs](https://github.com/debian-pi/raspbian-ua-netinst#installer-customization) explain this

>The post‑install.txt is executed at the very end of the installation process and you can use it to tweak and finalize your automatic installation

willemdh, 2016-04-07:
I'm sorry if I'm not seeing it. I can see installer-config.txt and the changes that could be done to install some extra packages for example, but how can I make it launch a Bash script exactly?

willemdh, 2016-04-07:
> The configuration files are read in as shell scripts, so you can abuse that fact if you so want to.

So if I just add postinstall.sh this wel get executed right?

mattotodd, 2016-04-07:
its going to be looking for a file called post‑install.txt , but its processed as a bash shell script (my guess is to slightly conceal the fact that you can do anything to the machine because you are running as root)

>The configuration files are read in as shell scripts, so you can abuse that fact if you so want to

willemdh, 2016-04-07:
Ok, thanks. Just did my first unattended install, which was very successful. Wil give a few more tries and test if I can download and execute a Bash script.

Mausy5043, 2016-04-07:
@mattotodd @willemdh : Actually `post-install.txt` is executed as a `busybox sh`ell script. Any fancy `bash`isms you may wish to put in the `post-install.txt` will very likely not work. 

And the same goes for the `installer-config.txt` which is executed at the start of the install. But because the system is not yet installed at that point this should really only be used to set some installer parameters as outlined in the `README.md`.

willemdh, 2016-04-07:
@Mausy5043 Thanks for clarifying. So is there some way to execute a 'normal' Bash script post installation?

Mausy5043, 2016-04-07:
No. `bash` is not available in the installer.

Mausy5043, 2016-04-07:
If you can wait until after the reboot, You might want to add a cronjob or init.d script

Mausy5043, 2016-04-07:
Why do you think you need `bash` ?

goranche, 2016-04-07:
well, actually, you can copy over your script to the installed system and execute it using bash in that environment... I'd first bind mount `/proc`, `/sys` and `/dev`, though...

goranche, 2016-04-07:
but that will have to be done from `post-install.txt`, of course

willemdh, 2016-04-07:
@Mausy5043 Well I would like to do a bunch of stuff:
- Download the latest version of my motd generator (https://github.com/willemdh/generate_motd) , put it in /etc/profile.d
- Install and configure an nrpe agent so it gets automatically monitored by Nagios

So how can I put a script which launches at boot in /etc/profile.d?

goranche, 2016-04-07:
you should be able to do all that using `ash` (that's the `sh` that comes with `busybox`) in `post-install.txt`

willemdh, 2016-04-07:
@goranche Ok, never heard of ash, will have to look into that. As I'm ok working with Bash it might be easier for me if I find a way to launch a Bash script after the reboot like @Mausy5043 suggested.
So any hint on how I can make the installer add this script to /et/profile.d?

Mausy5043, 2016-04-07:
In the v1.1.x branch we have provisions in place to help you accomplish this. Assuming that you are using the v1.0.8.1 release you could put the files in a directory on the SD-card and copy them over in `post-install.txt` something like this (untested):

```
mkdir -p /rootfs/etc/profile.d
cp /bootfs/motd/* /rootfs/etc/profile.d
```
Assuming you put the files on the SD-card in a directory called `motd`.

goranche, 2016-04-07:
well, first, I doubt that `/etc/profile.d` is the right way to go (to run a script at boot), since that will require a user to log into the system, before the script gets run... if you really want to go that way, use `/etc/rc.local`. either copy your script to there, or add an invocation to it... just remember to remove it after the first boot...

you could do a simple  `echo "echo 'first boot' > /bootfs/etc/rc.local` in post-install.txt... this will output "first boot" on **every** boot (since it's not being removed)... just as an example, of course :wink: 

regarding ash (the busybox sh)... it's a shell, nothing too special, from a script perspective it'll be quite close to bash, if you don't use the advanced features of bash...

willemdh, 2016-04-07:
@goranche Thanks for the tips! I'll do some tests with this and will update this thread the moment I got it working.

goranche, 2016-04-07:
you might want to take a look [here](https://github.com/goranche/raspbian-ua-netinst-ro/blob/master/post-install.txt)... it's a very simple `post-install.txt`, but you might get a feeling on how to do this...

if you want one that's "a bit" more complicated, look at [this](https://github.com/webcampsi/info-beamer/blob/master/raspbian-ua-netinst/post-install.txt)... :innocent:

willemdh, 2016-04-07:
Can I use curl or wget in post-install.txt?

goranche, 2016-04-07:
you can use wget in the installer, but IIRC there is no https support... but you can also do the following:
```chroot /rootfs wget https://www.somewhere.over.the.rainbow/some_file.txt -O /tmp/some_file.txt```

and after that the file will be available at `/rootfs/tmp/some_file.txt`

I hope the pattern is clear by now, `post-install.txt` runs with a separate filesystem, but the installed system is available under `/rootfs`... (I guess this could be better documented :thought_balloon: )

willemdh, 2016-04-09:
Ok, closing this up. I ended up with creating a cronjob in post-install.txt

    chroot /rootfs /bin/cp /boot/cron_raspbian_first_boot /etc/cron.d/naf_raspbian_first_boot

Which will execute a Bash script

    @reboot root naf_raspbian_config.sh

And do the things I need to do. Got another question, but I'll make a new issue for that. Thanks all. This was a fun learning experience.

closed 2016-04-09 by willemdh

Mausy5043, 2016-04-09:
Me thinks: `cp /bootfs/cron_raspbian_first_boot /rootfs/etc/cron.d/naf_raspbian_first_boot` might work better?

goranche, 2016-04-09:
I wouldn't say better, but easier, sure :+1:

willemdh, 2016-04-09:
My cron file like somehow is no longer doing it's job. the file gets created with:

    @reboot root /boot/naf_raspbian_config.sh

in it. Script permissions;

    ls -la /boot/naf_raspbian_config.sh
    -rwxr-xr-x 1 root root 3513 Apr  9 14:02 /boot/naf_raspbian_config.sh

And the cron file:

    ls -la /etc/cron.d/naf_raspbian_first_boot
-rwxr-xr-x 1 root root 83 Apr  9 14:57 /etc/cron.d/naf_raspbian_first_boot

No sure what's going on.. Any idea?

goranche, 2016-04-09:
not really, no... and like I commented before, it's not the way I'd do things... ;)
I'd just do things in post-install.txt using chroot

goranche, 2016-04-09:
as is obvious from the links I provided... :innocent:

diederikdehaas, 2016-04-09:
https://raspberrypi.stackexchange.com/a/39222 gives some tips on how to debug cron scripts

willemdh, 2016-04-10:
Well I finally went for the script in /etc/profile.d => This enables me to set some interactive settings which require user input.

I finished my MotD generator (https://github.com/willemdh/generate_motd) so it supports Raspbian now by the way. I should make some time to make a Raspberry theme one day. 

[image]

Mausy5043, 2016-04-10:
👍