debian-pi/raspbian-ua-netinst #400: How to achieve 15 minute install?

willemdh, 2016-04-09:
> installation takes about 15 minutes with fast Internet from power on to sshd running

I've done 10+ installs on RaspBarry Pi 2 with very minor modifications to the base install. I'm on high speed fibre 115 Mbps download.

But my installation from to start to being able to sshd still takes  more then 30 minutes it seems. Am I missing something?

goranche, 2016-04-09:
I'd say that with the newer debian, 15 minutes is highly unlikely... there is just much more that needs to be done...
also, it might be worth checking your SD card, because that will have the largest impact on the install speed

Mausy5043, 2016-04-09:
I think the 15 minutes is for the `base` install. But, even that could be off as the time mentioned in the docs was observed many moons ago.

willemdh, 2016-04-09:
Ok, thanks.

closed 2016-04-09 by willemdh

diederikdehaas, 2016-04-10:
> it might be worth checking your SD card

Both the size and the speed are VERY influential.
It's now been a while, but I regularly do `server` installs in 15-20 minutes, but I do use quality SD cards of >=16GB.

willemdh, 2016-04-10:
Using a Sandisk MicroSD Ultra Android 32GB 80MB/s Class 10 at the moment. How fast is yours if I may ask?

diederikdehaas, 2016-04-10:
I use Samsung EVO 16/32GB Class 10/UHS-1. The 80MB/s doesn't say much as it is all about random read-write speed, but nevertheless I would've expected better speeds with your card.

Cheap chinese and/or Transcend card are often problematic, speed wise.