debian-pi/raspbian-ua-netinst #403: startx

willemdh, 2016-04-25:
Just a small question before I start wasting time looking for something that might not be there. I've been working with ssh from the start. So today I tried accessing X, but when I type startx , I get command not found. So is there supposed to be a gui in the unattended installer or not?

Mausy5043, 2016-04-25:
The installed system comes without a GUI. X is not installed.

goranche, 2016-04-25:
no, none... you can install it yourself, of course, using `apt-get`, but you'll have to "install it all", X, a window manager, a desktop, applications, ... you might want to start searching for "debian install xfce" or similar (replace xfce with an environment of choice)

willemdh, 2016-04-25:
Ok, thanks for the superfast answer. I thought I broke X somehow. Not sure yet if I really want a gui yet. I'll have a look at xfce.

closed 2016-04-25 by willemdh