debian-pi/raspbian-ua-netinst #404: Question - Smallest possible image with networking

aristosv, 2016-05-06:
This question is driven by pure curiosity, so don't get me wrong.

A few days ago I set up a small surveillance system with 4 RaspberryPi 3's and MotionEyeOS. I was impressed to see that the MotionEyeOS image was a mere 50MB download. This included Networking,  SSH and Web access, even the Wi-Fi was working. 

Such a small image size, with networking enabled is the perfect base for many RPi projects. If you have networking, and apt-get, you can then install whatever's needed. 

So this got me thinking. Why can't I make the Raspbian image this small, using raspbian-ua-netinst? The best I could do so far is a few hundred megabytes.

So is it possible to create a tiny 50MB image with networking?

goranche, 2016-05-07:
it would be possible, of course, but you'd have to start from scratch and not go with debian (or pretty much any of the "standard" distributions)... you might achieve sizes close to that manually unpacking packages and copying over just the files you actually need, but it would be tedious work...

In a sense, the installer itself is such a system, limited in functionality, but a working system with networking nevertheless... and it clocks in at around 5MB (the cpio file containing the installer filesystem itself), so it's doable... 😉 

so, I guess I'm sorry to say you'll have a hard time achieving this goal with this installer...

goranche, 2016-05-07:
I've had something similar in mind, a small enough system that could load a (possibly compressed) filesystem to RAM and run it from there, completely from memory (of course, on 1GB RAM boards)...
but time constraints don't really allow for this exercise 😇  

finding time to spend on this project is hard enough as is 😒

diederikdehaas, 2016-05-07:
The minimum config I've used was based on the following:
```
preset=base
packages=dosfstools,net-tools,ifupdown # comma separated list of extra packages
```
You could probably leave out `dosfstools` and add a ssh server package, like `openssh-server` or more minimal `dropbear`.
That would create a normal system, stripped to pretty much the absolute minimum while still providing networking and ssh access. 
But I'm pretty sure you'd still end up with a couple of hundred MB, uncompressed. 

You can trim it down further, but then it does become tedious and very manual and would not be a normal system. You could remove all the man pages, localization files and documentation that come with a package. You can even decide to remove some executables which you (think) don't need.
Like @goranche said, that is what we do for the installer itself, but I won't recommend it for normal use.

closed 2016-10-03 by aristosv