debian-pi/raspbian-ua-netinst #405: Firmware update

willemdh, 2016-05-06:
Anyone has experience with rpi-update in combination with the unattended installer? As after the firmware upgrade (from 3.18.0-trunk-rpi2 to 4.xx) I try to reboot and my Pi hangs on boot apparently..

Mausy5043, 2016-05-07:
In `/boot/config.txt` you need to comment out these lines:
```
kernel=vmlinuz-3.18.0-trunk-rpi
initramfs initrd.img-3.18.0-trunk-rpi followkernel
```

<hr><hr>

To "upgrade" to a newer kernel I use this script: 
```bash
#!/bin/bash

# update the kernel to 4.4.8
# Linux rbian 4.4.8+ #880 Fri Apr 22 21:27:42 BST 2016 armv6l GNU/Linux
# tested stable commit:
# https://github.com/Hexxeh/rpi-firmware/commit/6d158adcc0cfa03afa17665715706e6e5f0750d2
echo "%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%"
echo "% THIS WILL UPDATE THE KERNEL %"
echo "%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%"
echo ""

echo ">>> 1: Update the apt-get cache"
sudo apt-get update

echo ""
echo ">>> 2: Install rpi-update"
sudo apt-get -yuV install rpi-update

pushd $HOME
  echo ""
  echo ">>> 3: Execute rpi-update"
  sudo rpi-update 6d158adcc0cfa03afa17665715706e6e5f0750d2

  echo ""
  echo ">>> 4: Adjust config.txt"
  sudo sed -i 's/kernel=/#kernel=/' /boot/config.txt
  sudo sed -i 's/initramfs/#initramfs/' /boot/config.txt
  cat /boot/config.txt
popd

echo ""
echo "%%%%%%%%%%%%%%%%%%%%%%%$%%%%%%%%"
echo "% REBOOT TO USE THE NEW KERNEL %"
echo "%%%%%%%%%%%%%%%%%%%%%%%%$%%%%%%%"
echo ""
```

The long string of numbers and letters is the commit hash. You can find one [here]( https://github.com/Hexxeh/rpi-firmware/commits/master). Best is to look for a commit with the title "Kernel: Bump to ...".  Those tend to be the stable kernels. At the right there is an icon that lets you **Copy the full SHA**. Paste it in place of the `6d158...` in the script and your good to go.
You may be required to enter the root password in order to execute the `sudo` lines.

willemdh, 2016-05-07:
Thanks a lot @Mausy5043 for this nice explanation. Didn't realize the latest firmwares were so unstable.
So I tried with https://github.com/Hexxeh/rpi-firmware/commit/6d158adcc0cfa03afa17665715706e6e5f0750d2 but same issue. This is the same commit you referenced in your script. My RaspBerry Pi 2 B hangs on the Multicolored start screen and has a red light. I'll try again with an older commit.
Are you 100 % sure you got firmware with commit 6d158adcc0cfa03afa17665715706e6e5f0750d2 working or was this just as an example?

willemdh, 2016-05-07:
J

willemdh, 2016-05-07:
Just one more question. Is this something that can be done in post-install.txt?

Mausy5043, 2016-05-07:
Not easily.
`rpi-update` requires that you input a "y" or "n" before it will start the upgrade.

willemdh, 2016-05-17:
Ok, thanks for the info. Closing up!

closed 2016-05-17 by willemdh