debian-pi/raspbian-ua-netinst #407: outdated documentation in README.md

oupala, 2016-05-27:
In README.md, section [Reinstalling or replacing an existing system](https://github.com/debian-pi/raspbian-ua-netinst/blob/master/README.md#reinstalling-or-replacing-an-existing-system), the documentation is outdated:

> Make sure you still have kernel_install.img and installer.cpio.gz in your /boot partition.

I have some rpi1 and rpi2 versionned files:

```bash
/boot>ls kernel*
kernel.img  kernel-rpi1_install.img  kernel-rpi2_install.img
/boot>ls installer*
installer-rpi1.cpio.gz  installer-rpi2.cpio.gz
```

Documentation should be updated.

Mausy5043, 2016-05-27:
Oops.

Mausy5043, 2016-05-27:
Thank you for bringing this to our attention.
We'll look into it.
If you wish you are welcome to contribute. The guidelines for contributing are [here](https://github.com/debian-pi/raspbian-ua-netinst/blob/master/CONTRIBUTING.md).

oupala, 2016-05-27:
Well, by reporting this issue, I **am** contributing.

I would have created a pull request if only I would know how to update the documentation.

Next time I have anything to report, and to fix by myself, I'll create a pull request.

Mausy5043, 2016-05-28:
👍

thijstriemstra, 2016-06-18:
@oupala try https://github.com/debian-pi/raspbian-ua-netinst/edit/master/README.md, edit and save.

oupala, 2016-06-19:
I was not clear in my previous message. In fact, I know how to edit a file with github. My problem is that I don't know what to change in the README file.

Mausy5043, 2016-06-20:
Had some time to spare so I threw something together.

diederikdehaas, 2016-08-10:
The change has now been merged, thus closing the issue. Thanks for the report :-)

closed 2016-08-10 by diederikdehaas