debian-pi/raspbian-ua-netinst #411: How post-install.txt work?

wannadrunk, 2016-06-21:
I'm just the basic level of Linux administration. I come across to this wonderful installer because I am going to setup quite a number of RPi.

I don't understand how the post‑install.txt work. There is no explantion in the document. And I assume it is just some shell script. I have put below code, try to update some system setting, but it doesn't work with error message "not found"

Anyone please guide me some light.

``` post-install.txt
#!/bin/sh

# System setting
/usr/bin/timedatectl set-timezone Asia/Bangkok
/usr/sbin/update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
```

``` raspbian-ua-netinst.log
=== Start executing post-install.txt. ===
/etc/init.d/rcS: /bootfs/post-install.txt: line 2: timedatectl: not found
/etc/init.d/rcS: /bootfs/post-install.txt: line 3: update-locale: not found
=== Finished executing post-install.txt. ===
```

wannadrunk, 2016-06-21:
I think I got the answer/solution for my post-install.txt. I will try it later.

Help with post-install.txt https://github.com/debian-pi/raspbian-ua-netinst/issues/374

diederikdehaas, 2016-06-21:
The main thing is you need to `chroot` into `/rootfs/` and then run your commands.

wannadrunk, 2016-06-21:
Revised post-install.txt and it works fine.
```
#!/bin/sh

# System setting
chroot /rootfs /usr/bin/timedatectl set-timezone Asia/Bangkok
chroot /rootfs /usr/sbin/locale-gen en_US-UTF-8 th_TH.UTF-8
chroot /rootfs /usr/sbin/update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
```
Thanks a lot for the great work.

closed 2016-06-21 by wannadrunk