debian-pi/raspbian-ua-netinst #416: Out of space! 512MB netinstall raspbian.

mlawtk, 2016-08-06:
I tried to install the latest image on a Pi  I with 512MB SD card.

It fails falling into an apt loop complaining no space left on device.

I think the latest image is probably too heavy, and perhaps some packages could be removed to make a working minimal install.

Another possibility is the size of SD Partitions is not optimal.

All I need at the start is apt and networking, and ssh.

Please advise to which image last worked on 512MB(or 480MB) memory card as was claimed in documentation.

Thanks

diederikdehaas, 2016-08-06:
The doc say: `can fit on 512MB SD`
But in order to still get that, you'd have to change the size of the boot partition and you'd have to use sth like `preset=base`.
I haven't tried lately, but I still think it **can** be done, but don't expect much of a useful system though.

Mausy5043, 2016-08-07:
On most of my Pies I see this:
```
Filesystem                       Size  Used Avail Use% Mounted on
/dev/mmcblk0p1                   128M   70M   59M  55% /boot
```

So, I guess a boot partition size of 80MB should be enough in most cases. The rest can then be assigned to `/`. As @diederikdehaas already said you may want to start experimenting with `preset=base` first.  The amount of space left on the root device also depends on what additional packages you install post-boot.

The 512MB is an extreme case. Personally, given their low price, I tend to use 2GB or 4GB cards. A "normal" install with some packages added requires just over 1GB (YMMV). E.g.:
```
Filesystem                       Size  Used Avail Use% Mounted on
/dev/root                        3.5G  1.1G  2.2G  33% /
```

mlawtk, 2016-08-07:
FYI, I am attempting install with cpio and kernel image of Pi II deleted (as Im using a Pi I).  This frees a few MB.  Awaiting results....

mlawtk, 2016-08-07:
SO... when the installer creates the filesystem it expands the 1st partition to 122MB, 2nd Partition to 374MB.  Any idea which file I need to edit in one of the precompiled images?  I want to make the first partition about 50MB giving second partition around 440MB which I hope would make these images work.

diederikdehaas, 2016-08-07:
`config.txt` has a parameter you can specify: `bootsize` with default value of `+128M` but you can change that ...

Mausy5043, 2016-08-22:
Issue #404 may be of help too.

Mausy5043, 2016-10-13:
@mlawtk : Has your problem been resolved. Or do you have additional questions/remarks?

Mausy5043, 2016-12-28:
No response from OP for three months. Therefore, assuming the issue is resolved and closing this issue.
If you feel the issue was closed in error, please feel free to re-open while adding additional info.

closed 2016-12-28 by Mausy5043