debian-pi/raspbian-ua-netinst #417: Corrupted image

richard-scott, 2016-08-06:
I'm building an image from v1.1.x and when I write it to my SD card (any SD Card I have multiple) the fat FS is not readable, and the SD card is not bootable. I have tried writing the image to the SD card from Windows, and from Linux using DD.

On Windows, it attempts to format the SD card when I insert it, and on Linux I am unable to mount the device as it has an unknown file system on it.

This never used to happen, but it has been a few months since I last attempted this. I'm 100% sure I'm writing out the same way I did before.

If I download an image from the "latest release" page, that works just fine.

Anyone else having the same problem?

richard-scott, 2016-08-06:
I've just tried "master" and that works, v1.0.x and that is fine too, its just v1.1.x that creates a corrupted image for me, I get this when trying to mount it:

```
[root@recon v1.1.x]# mount /dev/sdd1 /mnt/
mount: /dev/sdd1 is write-protected, mounting read-only
mount: wrong fs type, bad option, bad superblock on /dev/sdd1,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
```

mlawtk, 2016-08-06:
works for me using linux and xz image to flash with this command: 
xzcat ./Downloads/raspbian-ua-netinst-v1.0.8.1.img.xz > /dev/mmcblk0
Try this if you can.

diederikdehaas, 2016-08-06:
v1.0.x and master are (pretty much) equal, so it's no surprise that they both work.
I have never had a problem writing a v1.1.x image to SD card though.
How did you build the v1.1.x image? Please provide the exact steps/commands you used.

richard-scott, 2016-08-06:
yes, the downloaded ones work fine, the ones I build myself don't, but only for anything except v1.0.x.

I use this script to do my builds:

```
#!/bin/bash
GIT[${#GIT[@]}]="https://github.com/debian-pi/raspbian-ua-netinst.git v1.1.x debian-pi"
GIT[${#GIT[@]}]="https://github.com/debian-pi/raspbian-ua-netinst.git v1.0.x debian-pi"
GIT[${#GIT[@]}]="https://github.com/debian-pi/raspbian-ua-netinst.git master debian-pi"
#GIT[${#GIT[@]}]="https://github.com/Phiber2000/raspberrypi-ua-netinst.git master Phiber2000"
GIT[${#GIT[@]}]="https://github.com/FooDeas/raspberrypi-ua-netinst.git master FooDeas"
for (( i=0 ; i < ${#GIT[@]}; i++ )); do
  URL="$(echo ${GIT[${i}]} | awk  '{print $1}')"
  #echo "URL: ${URL}"
  BRANCH="$(echo ${GIT[${i}]} | awk  '{print $2}')"
  #echo "BRANCH: ${BRANCH}"
  DIR="/opt/raspbian-ua-netinst/$(echo ${GIT[${i}]} | awk  '{print $3}')"
  #echo "DIR: ${DIR}"
  #continue
  mkdir -p ${DIR}
  cd ${DIR}
  B=0
  if [ ! -d ${BRANCH} ]; then
    git clone -b ${BRANCH} --single-branch ${URL} ${BRANCH}
    B=1
  fi
  cd ${BRANCH}
  if [ ${B} -eq 0 ]; then
    if ! git pull 2>&1 | grep -q "Already up-to-date" ; then
      B=1
    fi
  fi
  if [ ${B} -gt 0 ]; then
    ./clean.sh
    ./update.sh
    ./build.sh
    ./buildroot.sh
  fi
done
```

The output folders are in /opt/raspbian-ua-netinst.

As you can see, nothing special takes place.

richard-scott, 2016-08-06:
I've uploaded the images I made to here: http://www.pointb.co.uk/raspbian-ua-netinst/

richard-scott, 2016-08-07:
I wrote this to test which image files are mountable:

```
#!/bin/bash
for I in $(find /opt/raspbian-ua-netinst -name raspbian*.img) ; do
  echo Image: "${I}"
  losetup --offset "$((2048*512))" -f ${I}
  L=$(losetup | grep ${I} | awk '{print $1}')
  echo "Loop: ${L}"
  if mount ${L} /mnt ; then
    umount /mnt
  fi
  losetup --detach ${L}
  echo ""
done
```

My output is as follows:

```
Image: /opt/raspbian-ua-netinst/debian-pi/v1.1.x/raspbian-ua-netinst-20160804-git39d7115.img
Loop: /dev/loop2
mount: /dev/loop2 is write-protected, mounting read-only
mount: wrong fs type, bad option, bad superblock on /dev/loop2,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.

Image: /opt/raspbian-ua-netinst/debian-pi/master/raspbian-ua-netinst-20160806-gite3bc3ab.img
Loop: /dev/loop2

Image: /opt/raspbian-ua-netinst/debian-pi/v1.0.x/raspbian-ua-netinst-20160806-git681da85.img
Loop: /dev/loop2
```

As you can see, some images are usable, and some are not.

richard-scott, 2016-08-07:
Interestingly enough, I've just moved my scripts onto a diff server and they work 100%. odd
Closing as: not a bug.

closed 2016-08-07 by richard-scott