debian-pi/raspbian-ua-netinst #422: Status of the v1.1.x branch compared to v1.0.x

bbinet, 2016-08-31:
I was wondering what is the status of the v1.1.x branch compared to v1.0.x?
What are the new features, enhancements of the v1.1.x branch?

Also, do you know when a new release will be published?

kpfleming, 2016-08-31:
I know one item: IPv6 support, which is on my agenda to complete this weekend.

BenjaminHae, 2016-08-31:
Support for the Pi Zero

bbinet, 2016-09-02:
Ok, so is it safe to start a new project based on the v1.1.x installer?

diederikdehaas, 2016-09-02:
> is it safe to start a new project based on the v1.1.x installer?

Yes. Keep in mind that new features can still be added to v1.1.x though.

bbinet, 2016-09-02:
Ok, thanks.

closed 2016-09-02 by bbinet

diederikdehaas, 2016-09-02:
For the most part v1.1.x is right now as it will be released though of which a first alpha should get out this month.