debian-pi/raspbian-ua-netinst #423: rasPI 3 stuck on: nonblocking pool is initialized

fredericofs, 2016-09-05:
installed raspbian-ua-netinst on a rasPI B+ model.
It is working ok.
But I cannot use this same SD card on a rasPI 3, or boot stucks with msg: nonblocking pool is initialized

kpfleming, 2016-09-05:
Which version of the installer did you use? Only the v1.1.x branch is compatible with RPi3, and in order to use it you'll need to build the installer image yourself (following the instructions in BUILD.md) since there aren't pre-built installers from that branch.

fredericofs, 2016-09-05:
Oh, I used the build from the download page, I see they are 1.0.X...
Is there any I can do with my actual installation to make it compatible with rasPI 3 ?

kpfleming, 2016-09-05:
Probably, but I suspect it's not easy. Unless you've put a lot of time into configuring it already, it would be easier to just do a new install using the v1.1.x branch. You'll end up with a better system anyway, as that branch installs a newer version of Raspbian (jessie), a newer kernel, and lots of other improvements.

bbinet, 2016-09-05:
Similar question: is it possible to build a unique SD card which can then be used to run both Pi1 and Pi3 devices?

closed 2016-09-05 by fredericofs

diederikdehaas, 2016-09-05:
> Is there any I can do with my actual installation to make it compatible with rasPI 3 ?

What you can try is adding a `[pi3]` block to `/boot/config.txt` with the same contents as the `[pi2]` block.
I'm pretty sure that would work as I've done it myself IIRC

@kpfleming 
> that branch installs a newer version of Raspbian (jessie), a newer kernel

The v1.0.x branch also installs jessie and the kernel that gets installed on the 'target' system (the one you boot after installation) is the same as with the v1.1.x branch.
I do agree that v1.1.x is more capable, mainly more settings you can configure during installation, and has some other improvements, like being able to freshly install on a RPi3 and a bootoffset of 8192, but what fredericofs tried to do should still work

diederikdehaas, 2016-09-05:
>  is it possible to build a unique SD card which can then be used to run both Pi1 and Pi3

I think the v1.1.x branch already does that now. In `/boot/config.txt` there is a section for `[pi1]`, `[pi2]` and `[pi3]` which gets 'executed' dependent on the detected RPi version.

bbinet, 2016-09-06:
@diederikdehaas in my case, I have only one `[pi3]` section in the `/boot/config.txt`, so it does not boot when I try to use the same SD card on a rpi1.
(I'm using an installed system based on `v1.1.x` branch).

diederikdehaas, 2016-09-06:
Then my guess is that you specified `hardware_versions` in your `installer-config.txt`, because otherwise a kernel + initramfs + corresponding sections in `/boot/config.txt` gets written for both RPi 1 and 2. As the RPi 3 uses the same kernel as the 2, copying the `[pi2]` to a `[pi3]` should be enough.

For the RPi 1 (and Zero) the `linux-image-rpi-rpfv` and for the RPi 2 & 3 the `linux-image-rpi2-rpfv` get installed. So making sure both meta-packages are installed should allow you to 'fix' your current system.

bbinet, 2016-09-06:
No, I haven't specified `hardware_versions` in my `installer-config.txt`.

Just to be sure we are on the same page, the original `config.txt` coming from the installer actually contains the `[pi1]`, `[pi2]`, `[pi3]` and `[pi0]` sections,  but I was talking about the installed system.
I made the installation on a pi 3, and now only the `[pi3]` section is available in `config.txt`, so the sd card does not boot anymore on all pi versions.

It would be great if by default we could use the same installed sd card on any pi version.

diederikdehaas, 2016-09-06:
> It would be great if by default we could use the same installed sd card on any pi version.

That _should_ be the default behavior and if it isn't that's sth to be investigated. I'll look into that.
When did you build the installer used to install the system?

bbinet, 2016-09-06:
I've built the installer out of the v1.1.x branch 4 days ago (2016/09/02)

kpfleming, 2016-09-06:
The current README.md in the v1.1.x branch says that 'detect' (the default) builds an installed system which is intended for the hardware used to do the installation, and that the user must set '1 2' to get an installed system which supports both RPi1 and RPi2/3.

bbinet, 2016-09-06:
@kpfleming thanks, I will try that.
Don't you think the '1 2' value should be the default?

diederikdehaas, 2016-09-06:
thanks for that @kpfleming :+1: 

It probably needs to be set to '1 2 3' though, but I'll have to go through the code to make sure.
This is then an unintended side effect for the code which supports RPi 3.

bbinet, 2016-09-07:
or even '0 1 2 3'?