debian-pi/raspbian-ua-netinst #426: locales option broken

ThomDietrich, 2016-09-14:
Hey guys, I was trying to set my locales and ended up with every locale property being set to `POSIX`.
(I am working with the latest from branch v1.1)
```
#installer-config.txt
locales=en_US.UTF-8 de_DE.UTF-8
system_default_locale=en_US.UTF-8
```
One hint I found are these lines in `/var/log/raspbian-ua-netinst.log`:
```
  locales =
  system_default_locale = en_US.UTF-8
```
Seems like the **locales string is not correctly copied over**?
Who can help making this work? Based on a PR I found I want to ping @goranche ;)

This is the content of my `installer-config.txt`.
Without these lines the default system state is described as follows:
```
$ locale
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE=de_DE.UTF-8
LC_NUMERIC="en_US.UTF-8"
..........="en_US.UTF-8"
LC_ALL=

$ locale -a
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_COLLATE to default locale: No such file or directory
C
C.UTF-8
POSIX
```

wannadrunk, 2016-09-14:
This might not be the solution. I update the locale and timezone via post-install.txt

`
#!/bin/sh

cat << 'EOF' > /rootfs/post-install.sh
#!/bin/sh

# System setting
## timezone
/bin/echo Asia/Bangkok > /etc/timezone
/bin/rm /etc/localtime
/bin/ln -s /usr/share/zoneinfo/Asia/Bangkok /etc/localtime

## locale
/bin/echo en_US.UTF-8 UTF-8 >> /etc/locale.gen
/bin/echo th_TH.UTF-8 UTF-8 >> /etc/locale.gen
/usr/sbin/locale-gen

EOF

chmod +x  /rootfs/post-install.sh
chroot /rootfs /post-install.sh
rm -f /rootfs/post-install.sh
`

ThomDietrich, 2016-09-14:
Hey @wannadrunk, thanks! I included your script lines in my setup script and it seems to work. I've additionally added at the end: 
`/usr/sbin/update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8 LANGUAGE=en_US.UTF-8`

I'll leave this Issue open as the original problem didn't get solved and might need to be fixed.

Mausy5043, 2016-09-15:
I must admit that I have no experience with the installer setting the locale.
I have this snippet in my `post-install.txt`

```
LANG="en_US"
CHARSET="UTF-8"
LOCALE=$LANG"."$CHARSET
LOCALEGEN=$LOCALE" "$CHARSET
echo -n "Setting locale : "$LOCALE
echo "LANG=$LOCALE" >> /rootfs/etc/default/locale
echo " ... $LOCALEGEN"
echo "$LOCALEGEN" > /rootfs/etc/locale.gen
chroot /rootfs /usr/sbin/locale-gen
```

Mausy5043, 2016-09-15:
It might be the space between the locales that's upsetting the installer. Could you try installing with this:

`locales="en_US.UTF-8 de_DE.UTF-8"`

ThomDietrich, 2016-10-04:
I'll have a look. Give me a few days.

diederikdehaas, 2016-10-04:
> It might be the space between the locales that's upsetting the installer.

Yep, quite sure that's the case. 
This is what I have in my `installer-config.txt`:
```
locales="en_US.UTF-8 nl_NL.UTF-8"
```

ThomDietrich, 2016-10-04:
Wonderful. Maybe you want to transfer that example to the example in README.md?
This issue may be closed.

closed 2016-10-13 by Mausy5043